team-vigir/flexbe_behavior_engine #1: Setting the flexbe_behaviors workspace folder

spmaniato, 2015-11-05:
Philipp, in the wiki/tutorial for installing the new FlexBE app you write:
> flexbe_behaviors folder should point to your newly created flexbe_behaviors **or equivalent ROS package**

I took that to mean that I can point `flexbe_behaviors` to `vigir_behaviors/vigir_flexbe_behaviors`. However, this is what I'm getting when roslaunching FlexBE:

```
process[behavior_launcher-5]: started with pid [28792]
[INFO] [WallTime: 1446734606.496283] Client connected.  1 clients total.
[INFO] [WallTime: 1446734606.625776] Using default behaviors package: flexbe_behaviors
[INFO] [WallTime: 1446734606.685261] [Client 0] Subscribed to /flexbe/behavior_update
Traceback (most recent call last):
  File "/home/cornell/vigir_repo/catkin_ws/src/flexbe_behavior_engine/flexbe_widget/bin/be_launcher", line 35, in <module>
    launcher = BehaviorLauncher()
  File "/home/cornell/vigir_repo/catkin_ws/src/flexbe_behavior_engine/flexbe_widget/src/flexbe_widget/behavior_launcher.py", line 29, in __init__
    manifest_folder = os.path.join(self._rp.get_path(behaviors_package), 'behaviors/')
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 199, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
rospkg.common.ResourceNotFound: flexbe_behaviors
```
And a second later the process dies:
```
[behavior_launcher-5] process has died [pid 28792, exit code 1, cmd /home/cornell/vigir_repo/catkin_ws/src/flexbe_behavior_engine/flexbe_widget/bin/be_launcher __name:=behavior_launcher __log:=/home/cornell/.ros/log/74db7392-83cb-11e5-83f6-3417ebd579fa/behavior_launcher-5.log].
log file: /home/cornell/.ros/log/74db7392-83cb-11e5-83f6-3417ebd579fa/behavior_launcher-5*.log
```

For completeness, I should also say that I've set the behaviors folder to `vigir_behaviors/behaviors` and the package namespace to `vigir`.

I'm puzzled by `Using default behaviors package: flexbe_behaviors`. Shouldn't it be using `vigir_flexbe_behaviors`? Is there a bug or am I doing something wrong? Thanks

pschillinger, 2015-11-05:
You need to set the rosparam `behaviors_package` to "vigir_flexbe_behaviors". See [behavior_onboard_atlas.launch](https://github.com/team-vigir/vigir_behaviors/blob/master/vigir_flexbe_launch/launch/behavior_onboard_atlas.launch#L7) for example.

I guess I missed to put it in the tutorial, will update it as soon as possible. Sorry!

spmaniato, 2015-11-05:
OK, thanks. But does this mean that I have to launch onboard as well any time I want to use the FlexBE Chrome app?

pschillinger, 2015-11-05:
Yes, you can add it to any launch file.

Would you prefer it to be an environment variable?

spmaniato, 2015-11-05:
I'm fine either way. As long as it's easy to set up.

I just added the `rosparam` to `behavior_ocs.launch` and it's not complaining now, so I'm happy :stuck_out_tongue:

rojas70, 2016-06-06:
has this been updated in the tutorial? should this issue be closed?

pschillinger, 2016-06-06:
Yes, I guess this can be closed.

Noted that it got changed recently, the launch files now accept an arg [behaviors_package](https://github.com/team-vigir/flexbe_behavior_engine/blob/master/flexbe_onboard/launch/behavior_onboard.launch#L4) to point to the desired folder.

I don't think this is mentioned anywhere in the tutorials yet (please correct me if I am wrong), but there should be an advanced tutorial covering this eventually.

closed 2016-06-06 by pschillinger