team-vigir/flexbe_behavior_engine #4: sudo required for create_repo

dcconner, 2016-03-14:
The create_repo script tries to install rosbridge, which requires super user privileges.

Can we just make rosbridge-server a run dependency, and not do this sudo.

Or can we just use `dpkg -s ros-jade-rosbridge-server` and check return type, and only try sudo if not installed?

skohlbr, 2016-03-15:
We use this approach to only run sudo if required:
https://github.com/tu-darmstadt-ros-pkg/centaur_robot_tutorial/blob/master/update.sh#L54
I think this could also be done in this case?

dcconner, 2016-03-15:
That works for my issue.  I can make the change, but want Philipp to weigh in first.

pschillinger, 2016-03-20:
Sounds good, feel free to integrate the change Stefan suggested or assign it back to me.

commit 492bab05 (2016-05-07): [flexbe_widget] Only require sudo in create_repo if pkg needs to be installed (resolve #4)
closed 2016-05-07 by pschillinger