team-vigir/flexbe_behavior_engine #5: create_repo defines separate git repo ... make optional?

dcconner, 2016-03-15:
I am creating some states that are grouped with other packages in a repo, so I want to add the behaviors to the this same repo.

The create_repo script assumes a separate repo, and does a "git init" to create the .git folder.
I deleted this for my setup, but this seems to be a big assumption on the scripts part.

Should we make this optional based on a command line argument?

pschillinger, 2016-03-20:
I don't completely understand. The purpose of create_repo is to create a repo. It sounds like you would prefer to have another script only doing a subset of this. If you tell me what exactly the script should do, I can add one.

dcconner, 2016-03-21:
The create_repo sets up the specific file structure expected of a behaviors package as well as creates the git repo initialization.  

I'm asking for a script to create the file structure and demo, without git initialization.

Also, as related to direct email, it would be nice to use custom package name for the demo so that you could run the script multiple times in a workspace.

pschillinger, 2016-03-21:
Alright, I will prepare such a script.

Achllle, 2020-01-30:
I was also surprised that a git repo is created since I created the packages inside an existing git repo. +1

simonbogh, 2022-02-24:
Was this ever added? It would be really great to have.

dcconner, 2023-07-27:
Newest ROS 2 version on https://github.com/FlexBE/flexbe_behavior_engine/blob/ros2-devel/flexbe_widget/bin/create_repo has more interactive mode.

closed 2023-07-27 by dcconner