team-vigir/flexbe_behavior_engine #8: Cancel behavior with be_action_server

icemanx, 2016-06-01:
Can we stop or cancel a running behavior with action server or any other way exists to accomplish this?

pschillinger, 2016-06-02:
A running behavior can always be stopped by sending a message of type `std_msgs/Empty` to the topic `/flexbe/command/preempt`.

Note that this can only stop the behavior executive. If a state started an external process like making an action call, it has to make sure that this process is properly stopped when stopping the behavior or leaving the state, which is typically done in the respective event callbacks. See the tutorial [State Lifecycle](http://wiki.ros.org/flexbe/Tutorials/The%20State%20Lifecycle) for more information.

Right now, cancelling the action call is not connected to stopping the behavior. But I think this should be added and I will do this as soon as I find time.

icemanx, 2016-06-02:
Thank you for giving me a perspective for cancelling behaviors. I just implemented and tested "fast-written" solution for action server and created a pull request. If you have suggestions, I'll be very glad to hear it.

pschillinger, 2016-06-02:
Looks perfectly fine, I will merge it. Thank you for your contribution!

commit 9d73bc64 (2016-06-02): Merge pull request #9 from icemanx/master

Added behavior stopping feature for behavior action server (resolve #8)
closed 2016-06-02 by pschillinger