team-vigir/flexbe_behavior_engine #14: Behavior Construction Failed

rojas70, 2016-06-06:
http://wiki.ros.org/flexbe/Tutorials/Execution%20of%20a%20Behavior

My attempt to "start the execution" of the behavior is failing. I get the following error message in the Behavior Feedback Window:

````
[12:13:49 AM] Onboard engine just started.
[12:14:10 AM] --> Starting new behavior...
[12:14:10 AM] Behavior construction failed! [-]
    'HelloWorldDemoSM' object has no attribute 'hello'
[12:14:10 AM] No behavior active.
[12:14:10 AM] Dropped behavior start request because preparation failed.
````

And in the terminal:
````
 roslaunch flexbe_onboard behavior_testing.launch 
... logging to /home/vmrguser/log/3551e4a4-2bf5-11e6-ab6f-e4b31826a796/roslaunch-baymax-2358.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://baymax.local:40105/

SUMMARY
========

PARAMETERS
 * /behavior/behaviors_package: flexbe_behaviors
 * /behavior_launcher/behaviors_package: flexbe_behaviors
 * /behavior_rosapi/namespace: 
 * /behavior_rosbridge/port: 9090
 * /rosdistro: indigo
 * /rosversion: 1.11.19

NODES
  /
    behavior (flexbe_onboard/start_behavior)
    behavior_flexbe (flexbe_widget/flexbe_app)
    behavior_input (flexbe_input/behavior_input)
    behavior_launcher (flexbe_widget/be_launcher)
    behavior_mirror (flexbe_mirror/behavior_mirror_sm)
    behavior_rosapi (rosapi/rosapi_node)
    behavior_rosbridge (rosbridge_server/rosbridge_websocket)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[behavior_mirror-1]: started with pid [2376]
process[behavior_input-2]: started with pid [2377]
process[behavior_rosbridge-3]: started with pid [2378]
process[behavior_rosapi-4]: started with pid [2379]
process[behavior_flexbe-5]: started with pid [2380]
/usr/bin/google-chrome
Using google-chrome for FlexBE ...
process[behavior_launcher-6]: started with pid [2383]
process[behavior-7]: started with pid [2389]
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
[INFO] [WallTime: 1465229628.932580] Rosbridge WebSocket server started on port 9090
Created new window in existing browser session.
[behavior_flexbe-5] process has finished cleanly
log file: /home/vmrguser/log/3551e4a4-2bf5-11e6-ab6f-e4b31826a796/behavior_flexbe-5*.log
[INFO] [WallTime: 1465229629.137251] Ready for data requests...
[INFO] [WallTime: 1465229629.138073] Rosapi started
[INFO] [WallTime: 1465229629.286178] Using custom behaviors package: flexbe_behaviors
[INFO] [WallTime: 1465229629.319975] Using custom behaviors package: flexbe_behaviors
[INFO] [WallTime: 1465229629.377979] Parsing available behaviors...
[INFO] [WallTime: 1465229629.402755] 2 behaviors available, ready for start request.
[INFO] [WallTime: 1465229629.425468] Client connected.  1 clients total.
[INFO] [WallTime: 1465229629.737595] [Client 0] Subscribed to /flexbe/behavior_update
[INFO] [WallTime: 1465229629.747553] [Client 0] Subscribed to /flexbe/outcome_request
[INFO] [WallTime: 1465229629.752873] [Client 0] Subscribed to /flexbe/log
[INFO] [WallTime: 1465229629.757888] [Client 0] Subscribed to /flexbe/status
[INFO] [WallTime: 1465229629.762813] [Client 0] Subscribed to /flexbe/command_feedback
[INFO] [WallTime: 1465229629.766943] [Client 0] Subscribed to /flexbe/heartbeat
[INFO] [WallTime: 1465229629.772064] [Client 0] Subscribed to /flexbe/uicommand
[INFO] [WallTime: 1465229629.887621] --- Behavior Engine ready! ---
[INFO] [WallTime: 1465229629.888059] No onboard behavior is active.
[INFO] [WallTime: 1465229629.888484] --- Behavior Mirror ready! ---
[INFO] [WallTime: 1465229650.739080] Request for behavior Hello World Demo
[INFO] [WallTime: 1465229650.739634] No changes to behavior version.
[INFO] [WallTime: 1465229650.739763] Received a new mirror structure for ID 196636834
[INFO] [WallTime: 1465229650.740079] --> Starting new behavior...
[INFO] [WallTime: 1465229650.825493] Successfully applied 0 modifications.
[INFO] [WallTime: 1465229650.867202] Behavior Hello World Demo created.
[INFO] [WallTime: 1465229650.867485] The following parameters will be used:
[INFO] [WallTime: 1465229650.867752] waiting_time = 1
[ERROR] [WallTime: 1465229650.870646] Error raised during SMACH container construction: 
Traceback (most recent call last):

  File "/home/vmrguser/ros/indigo/baxter_ws/src/birl_baxter/birl_executive/flexbe_behavior_engine/flexbe_onboard/tmp/tmp_196636834.py", line 67, in create
    LogState(text=self.hello, severity=Logger.REPORT_HINT),

AttributeError: 'HelloWorldDemoSM' object has no attribute 'hello'

[ERROR] [WallTime: 1465229650.871035] Behavior construction failed!
'HelloWorldDemoSM' object has no attribute 'hello'
[INFO] [WallTime: 1465229650.871642] No onboard behavior is active.
[INFO] [WallTime: 1465229650.871971] --- Behavior Mirror ready! ---
[ERROR] [WallTime: 1465229650.921742] Dropped behavior start request because preparation failed.
[INFO] [WallTime: 1465229650.922156] --- Behavior Engine ready! ---
```

However, if I go to the "Behavior Dashbord", under private configuration I have the variable:
`hello= "hello world"`.

I also have the following workspace setup, not sure what might be wrong. 
````
-src
|____birl_executive
         |____flexbe_behavior_engine
         |____flexbe_pa_demo_behaviors
                  |_____behaviors
                  |_____flexible_behaviors
                  |_____other_states
                             |_____flexbe_pa_demo_flexbe_states
                                        |_____CMakesList.txt
                                        |_____packages.xml
                                        |_____setup.py
                                        |_____flexbe_pa_demo_flexbe_states
                                                   |_____example_action_state.py
                                                   |_____example_state.py
                                                   |_____ __init__.py
                             |_____flexbe_pa_demo_flexbe_states
                             |_____flexbe_pa_demo_flexbe_states
````

pschillinger, 2016-06-06:
Workspace setup looks perfectly fine.

I think the error is caused by `self.hello` as the value for `text` in the log state. It should normally only say `hello`. The "self." refers to behavior parameters (or other instance variables), while `hello` in this case is a private variable.

rojas70, 2016-06-07:
Got it. I missed that part. It's easy to miss on the first go around as one is getting familiarized with the system.

Thanks!

closed 2016-06-07 by rojas70