team-vigir/flexbe_behavior_engine #16: No error is shown if a state is not loaded

daenny, 2016-06-22:
If you create a state in your flexbe states folder and there is some issue with it, it does not load, and there is also no error shown anywhere in the interface. 
I.e. create an example state like:
``` python
from flexbe_core import EventState, Logger

class ExampleState(EventState):
	'''
	This state has an output key, but no second string after the test. So it will not load.
	#> input_key test
	<= continue 			All done.
	'''

	def __init__(self):
		# Declare , input_keys, and output_keys by calling the super constructor with the corresponding arguments.
		super(ExampleState, self).__init__(outcomes = ['continue'])
		# Thus, we cannot save the starting time now and will do so later.
		self._start_time = None

	def execute(self, userdata):
		return 'continue'
```
This state will not load due to the line:
```
	#> input_key test
```
It expects a second item after test, i.e. this will work:
```
	#> input_key test description
```
But there is no error shown in the flexbe widget or anywhere else showing me that this is the case. So it took a while to find out the issue. There should be a warning or something that the state is not loaded and where it went wrong.

pschillinger, 2016-06-22:
Good point, will look at it.

Especially if it is just related to the documentation, it should not lead to the state being ignored. Instead, just the respective field should be undocumented and a warning will help, I agree.

pschillinger, 2016-06-28:
Should be fixed now (version 1.0.9) with [this commit](https://github.com/pschillinger/flexbe_chrome_app/commit/31a8588684c651ea4134f0d6abeb8104fb284b66).

Thanks again for reporting!

closed 2016-06-28 by pschillinger

daenny, 2016-06-28:
Thanks for the quick reply and resolution :)!