team-vigir/flexbe_behavior_engine #17: Concurrent states execute on_exit() twice

bkvaluemeal, 2016-07-05:
[image]
[files.tar.gz](https://github.com/team-vigir/flexbe_behavior_engine/files/348579/files.tar.gz)

bkvaluemeal, 2016-07-05:
The files I included are a modified log state changed from working on "on_enter()" to "on_exit()" and a behavior with a single concurrent container containing 3 of those modified states. When I run the behavior, each of the 3 log messages is printed out twice as shown in the screenshot. I don't believe that is the intended behavior.

pschillinger, 2016-07-06:
Thanks for reporting! I will look into that.

pschillinger, 2016-07-06:
The problem was that on_exit was called once for each particular state when it finished execution and once for all states of the container when it was left. If you would connect the outcomes of the three log states to different outcomes of the container, you would have seen that only one of them is executed twice.

Should be fixed now, on_exit is only called for states when leaving the container if they are still active.

commit 86e4ef8f (2016-07-06): [flexbe_core] Only call on_exit on cc leave for states which are still looping (fix #17)
closed 2016-07-06 by pschillinger