team-vigir/flexbe_behavior_engine #19: Configuration Profiles (Feature Request)

romayiii, 2016-10-07:
Hi!

Is there a plan on having FlexBe Configuration profiles?

I currently find myself working with three different robots, each one of them uses FlexBe, but I need to change the Flexbe configuration manually each time I want to work with a different robot.

I would be awesome to have a "Load/Save Profile" button in the Configuration Section ;)

dcconner, 2016-10-07:
There is a export/load configuration in the latest FlexBE app, but unfortunately the Java Script uses hash codes that are machine specific.  You can export the file to text editor and use that to copy and paste the paths into the other machine, but you cannot load the configuration directly

+2 on getting this to be completely machine independent.

pschillinger, 2016-10-07:
Yes, starting with version 1.0.7, there is a profile functionality. When you go to the Configuration page, there are two buttons (import, export) in the action bar at the top of the window.

The hashes (in fact, file entry IDs) come from Chrome and yes, they are machine specific, unfortunately. This is due to Chrome's security policy which requires an app to explicitly ask for permission before giving access to the respective part of the file system. As a consequence, these file entry IDs need to be different for each machine.

This has been discussed previously at the `flexbe_chrome_app` repo: [[Feature] Read configuration from file #3](https://github.com/pschillinger/flexbe_chrome_app/issues/3). I agree that it would be very useful, but such a feature will most likely not be possible before getting rid of Chrome (which is planned, but will need more time).

closed 2023-05-17 by dcconner