team-vigir/flexbe_behavior_engine #21: Live-View of userdata

fmauch, 2016-11-08:
When reviewing the execution of a larger behaviour it is quite usefull to be able to see the current userdata and it's contents, so it is easier to debug what's currently going on.

While it is already possible to see a state's userdata and it's remapping, the userdata's content is not exposed to the user in the control view.

pschillinger, 2016-11-08:
Yes, values of userdata are not being displayed. This would potentially
require a significant amount of data transmission and userdata can contain
arbitrary objects.

However, I agree that it would be very useful in some cases and especially
for debugging before deploying a system. I can very well imagine to add
something like an on-demand value display to the Runtime Control view of
the UI.

Anyone, feel free to post ideas or requests regarding realization so that I
can incorporate them.

fmauch, 2016-11-11:
publishing the userdata to a topic whenever a transition happens would be an option. This can be created only if a subscriber to this topic exists, so the transfer overhead would be optional to the user.

Also, publishing (or showing) a list of userdata keys and then some on-demand  display sounds like a good way to go.

dcconner, 2017-03-28:
I support on demand only publishing of userdata to a topic, preferably time stamped for data logging.

The question would be how to organize for greatest utility, without incurring significant overhead.  
Just logging user data as a char buffer would be simply, but not very useful with rqt_bag or other tools.
What python tools could we use to allow us to extract and post process?

Adding structure (e.g. via python pickle) would incur overhead, but at least we could read from bag file and extract data, even if we could not view in rqt_bag.

dcconner, 2023-05-17:
Should we add a state that will publish userdata on_enter then just return done ?

Possibly dump all by default, or only a specific key if specified?

dcconner, 2023-06-09:
Note that Noetic now has a get_userdata service.
We will soon push this to ros2-devel  branch at the new home https://github.com/flexbe/flexbe_behavior_engine