team-vigir/flexbe_behavior_engine #34: Pycharm can't resolve flexbe references

bennypi, 2017-06-26:
Hi, 

I've followed the installation steps on http://philserver.bplaced.net/fbe/download.php and ran the create_repo command. Although i executed this in the src-directory of my kinetic workspace, the command returned `The repository is not on the ROS_PACKAGE_PATH.
Run the following command if this was not on purpose:
  roscd && cd ../src && mv /home/benny/xenial_ws/src/calibration_controller_behaviors calibration_controller_behaviors`

However `echo $ROS_PACKAGE_PATH` returns `/home/benny/xenial_ws/src/calibration_controller_behaviors/behaviors/behavior_example_behavior:/home/benny/xenial_ws/src/calibration_controller_behaviors/behaviors/behavior_hello_world_demo:/home/benny/xenial_ws/src/calibration_controller_behaviors/calibration_controller_flexbe_states:/home/benny/xenial_ws/src/rac_rosnodes/caltab_detector:/home/benny/xenial_ws/src/calibration_controller_behaviors/flexbe_behaviors:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_core:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_input:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_mirror:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_msgs:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_onboard:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_states:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_testing:/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_widget:/home/benny/xenial_ws/src/robot_assisted_calibration:/home/benny/xenial_ws/src/universal_robot/universal_robot:/home/benny/xenial_ws/src/universal_robot/ur_bringup:/home/benny/xenial_ws/src/universal_robot/ur_description:/home/benny/xenial_ws/src/universal_robot/ur10_moveit_config:/home/benny/xenial_ws/src/universal_robot/ur3_moveit_config:/home/benny/xenial_ws/src/universal_robot/ur5_moveit_config:/home/benny/xenial_ws/src/universal_robot/ur_driver:/home/benny/xenial_ws/src/universal_robot/ur_gazebo:/home/benny/xenial_ws/src/universal_robot/ur_kinematics:/home/benny/xenial_ws/src/universal_robot/ur_msgs:/home/benny/xenial_ws/src/video_stream_opencv:/opt/ros/kinetic/share`

I ran successfully the hello world behaviour inside of the chrome app as well as from the console.
But when I import the package `calibration_controller_flexbe_states` into pycharm and open the file `example_action_state.py`, pycharm can't resolve e.g. flexbe_core or flexbe_core.proxy, but other ros imports like rospy are working. 

I would prefer creating flexbe states with IDE support, so maybe you can help me using flexbe in pycharm.

Greetings
Benny

dcconner, 2017-06-26:
Are you running PyCharm from terminal or from GUI launcher?   If latter, try invoking PyCharm from the command line of terminal with proper package setup (i.e. terminal where you can "roscd" into the flexbe states package).

bennypi, 2017-06-26:
I'm running it from terminal in which I sourced the xenial_ws/devel/setup.bash before starting pycharm. `roscd flexbe_core` changes the directory to `xenial_ws/src/flexbe_behavior_engine/flexbe_core`. 

When I try to open the declaration of `import flexbe_core` pycharm opens the file `/home/benny/xenial_ws/devel/lib/python2.7/dist-packages/flexbe_core/__init__.py` with the following content:
```python
# -*- coding: utf-8 -*-
# generated from catkin/cmake/template/__init__.py.in
# keep symbol table as clean as possible by deleting all unnecessary symbols

from os import path as os_path
from sys import path as sys_path

from pkgutil import extend_path

__extended_path = "/home/benny/xenial_ws/src/flexbe_behavior_engine/flexbe_core/src".split(";")
for p in reversed(__extended_path):
    sys_path.insert(0, p)
    del p
del sys_path

__path__ = extend_path(__path__, __name__)
del extend_path

__execfiles = []
for p in __extended_path:
    src_init_file = os_path.join(p, __name__ + '.py')
    if os_path.isfile(src_init_file):
        __execfiles.append(src_init_file)
    else:
        src_init_file = os_path.join(p, __name__, '__init__.py')
        if os_path.isfile(src_init_file):
            __execfiles.append(src_init_file)
    del src_init_file
    del p
del os_path
del __extended_path

for __execfile in __execfiles:
    with open(__execfile, 'r') as __fh:
        exec(__fh.read())
    del __fh
    del __execfile
del __execfiles
```

Thanks four your help.

pschillinger, 2017-06-26:
I can reproduce the observation when running PyCharm from a ROS terminal.

Although I am not a PyCharm user myself, it looks like this is because of the way catkin organizes Python packages in the devel space. PyCharm parses the `__init__.py` files in order to find modules, classes, etc, but this does not work with the way catkin stores the `__init__.py` (for dynamically loading modules from the src space).

For me, PyCharm recognizes any installed package (these` __init__.py` files are normal) and message definitions from the workspace (e.g., `from flexbe_msgs.msg import BehaviorLog` works for me). But any other package from the catkin workspace using a setup.py is highlighted as error.

I guess a way to add the src space packages directly to the Python path could fix this, but I don't know if there is a way to do so directly within PyCharm.

bennypi, 2017-06-26:
Actually there is a way to do this in PyCharm, and this also solves the problem for me! :)

This is how you do it:
1. Open PyCharm's settings
2. Open your project on the left of the settings window
3. Open Project Structure
4. Select the flexbe_states project
5. Click on Add Content Root
6. Select `xenials_ws/src/flexbe_behavior_engine
7. Mark this folder as source
8. Mark the subdirectory flexbe_core as source
9. Mark the subdirectory flexbe_core/src as source
10. Click on OK and wait until PyCharms triggers run and find the flexbe sources. 

Now autocompletion works for flexbe_core imports.

Thank you guys!

pschillinger, 2017-06-26:
Glad to hear that it worked!

You might probably want to post these instructions on ROS Answers as David pointed out in order to let others know how to do this.

closed 2017-06-26 by pschillinger