team-vigir/flexbe_behavior_engine #35: Onboard Status: disconnected

linhongbin, 2017-08-14:
Hi,

I am new user of Flexbe. After I installed Flex and followed the tutorial, I was stuck when I use Runtime Control panel of Flexbe App. The Onboard Status stay disconnected and therefore cannot execute the behavior.
[image]

I will appreciate it if you can offer a solution.

Ben.

linhongbin, 2017-08-14:
PS, I have run the commands:
roslaunch flexbe_onboard behavior_onboard.launch
rosrun rosbridge_server rosbridge_websocket

But the onboard status still showed disconnnected

linhongbin, 2017-08-14:
Similar question was post on ROS Q&A:
https://answers.ros.org/question/248690/flexbe-does-but-does-not-connect/

dcconner, 2017-08-14:
Look at closed issue #28

If that doesn't answer you question, then post you exact startup commands in all terminals, and post a "rostopic list"  output.  I believe you are missing something in your setup.   I have moved on to using the "FlexBE App" (See #29) to avoid issues with RosBridge and Chrome.

linhongbin, 2017-08-15:
Thanks a lot, the answer on #28 really helps me. 
It turns out the rosbridge didn't install properly in the system. 
After I fixed some dependency problem and sudo apt-get install ros-indigo-rosbridge-server, it works again. Thanks for your quick reply.

closed 2017-08-17 by dcconner

julianhfmn, 2023-04-04:
Hi, 

I am new to FlexBe as well and have the same or a similar problem.
[image]

I have the same error message but get asked to use `roslaunch flexbe_app flexbe_ocs.launch no_app:=true` instead of `rosrun rosbridge_server rosbridge_websocket`.

I start the app by using `roslaunch flexbe_app flexbe_full.launch` as specified in this [tutorial](http://wiki.ros.org/flexbe/Tutorials/Execution%20of%20a%20Behavior). 

The onboard behavior is disconnected, and when I use the mentioned `roslaunch` calls I get:

For `roslaunch flexbe_onboard behavior_onboard.launch`:
```
SUMMARY
========

PARAMETERS
 * /behavior/enable_clear_imports: False
 * /behavior/log_enabled: False
 * /behavior/log_folder: ~/.flexbe_logs
 * /behavior/log_level: INFO
 * /behavior/log_serialize: yaml
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    behavior (flexbe_onboard/start_behavior)

auto-starting new master
process[master]: started with pid [178868]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ffcbbb92-d2f1-11ed-964d-ab860ca1dc3c
process[rosout-1]: started with pid [178885]
started core service [/rosout]
process[behavior-2]: started with pid [178892]
[INFO] [1680617236.233025]: --- Behavior Engine ready! ---
shutdown request: [/behavior] Reason: new node registered with same name
[behavior-2] process has finished cleanly
log file: /home/julian/.ros/log/ffcbbb92-d2f1-11ed-964d-ab860ca1dc3c/behavior-2*.log
[behavior-2] restarting process
process[behavior-2]: started with pid [179869]
shutdown request: [/behavior] Reason: new node registered with same name
```
This continues in a loop where the terminals launch nodes with the same name, leading the other terminal to shut down the node and restarting it somehow.

For `roslaunch flexbe_app flexbe_ocs.launch no_app:=true`:
```
SUMMARY
========

PARAMETERS
 * /behavior/enable_clear_imports: False
 * /behavior/log_enabled: False
 * /behavior/log_folder: ~/.flexbe_logs
 * /behavior/log_level: INFO
 * /behavior/log_serialize: yaml
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    behavior (flexbe_onboard/start_behavior)

ROS_MASTER_URI=http://localhost:11311

process[behavior-1]: started with pid [171301]
[INFO] [1680615681.007665]: --- Behavior Engine ready! ---
```

The output of `rostopic list`:
```
/flexbe/behavior_update
/flexbe/command/attach
/flexbe/command/autonomy
/flexbe/command/lock
/flexbe/command/pause
/flexbe/command/preempt
/flexbe/command/repeat
/flexbe/command/sync
/flexbe/command/transition
/flexbe/command/unlock
/flexbe/command_feedback
/flexbe/heartbeat
/flexbe/log
/flexbe/mirror/outcome
/flexbe/mirror/preempt
/flexbe/mirror/structure
/flexbe/mirror/sync
/flexbe/outcome_request
/flexbe/request_behavior
/flexbe/request_mirror_structure
/flexbe/start_behavior
/flexbe/status
/flexbe/ui_version
/flexbe/uicommand
/flexbe/uinotification
/rosout
/rosout_agg
```

How can I fix this? I installed FlexBE both from source and as binary, nothing improved. Right now it is from source.
I took a look at #28 but as I am not asked by the app to launch the rosbridge I am not sure how much this applies to this version of the app. I still reinstalled ros-noetic-rosbridge-server, nothing changed.

Thank you for your help!