team-vigir/flexbe_behavior_engine #36: States with long execution times lead to sequential execution in concurrency containers.

mbharatheesha, 2017-08-16:
We (@mklomp, @gavanderhoorn and me) have been experimenting with the concurrency containers and observed the following behavior:

The states (or state machines) within a concurrency container are all started at the same time. However, if one of the states (whether the main state or otherwise) involves a time intensive computation (or just a long wait), all other concurrent states are blocked. The blocked states eventually do end up finishing, but only in a sequential manner.

Is this expected?

dcconner, 2017-08-16:
Without looking at the code, I believe the intent is that each state  in concurrent container gets one call of execute each time that the concurrent execute is called.  This is consistent with your observed behavior.   Generally we don't want to put long running calcs in the states.  We normally do that in a separate node and use a ROS ActionLib interface.  Then you return based on the Action result.   This doesn't block.

pschillinger, 2017-08-16:
Yes, the observation you describe is indeed to be expected. This is for example mentioned in [State Design Discussions](http://wiki.ros.org/flexbe/Tutorials/State%20Design%20Discussions):

> In general, state execution is assumed to be non-blocking. Main reason for this is the ability to let a remote operator interact with the behavior and intercept at any point during runtime. Thus, make sure to not block execution with any computationally intensive calculations.

In fact, a concurrency container checks the `execute` functions once a "tick" for each state. See for example the [WaitState](https://github.com/team-vigir/flexbe_behavior_engine/blob/master/flexbe_states/src/flexbe_states/wait_state.py#L28) or the tutorial [Developing Actionlib States](http://wiki.ros.org/flexbe/Tutorials/Developing%20Actionlib%20States) for example patterns.

mbharatheesha, 2017-08-17:
@pschillinger and @dcconner Thanks for the clarification. Missed that bit from the State Design Discussions part.

closed 2017-08-17 by dcconner