team-vigir/flexbe_behavior_engine #38: Remove the requirement for behavior default value override

tcassell94, 2017-09-26:
Let's say I have a behavior named BehA, which has a state within it that requires a userdata input key named KeyA. The default value for KeyA is stored in the State Machine Userdata box of BehA. 

Sometimes I want to run BehA standalone, and sometimes I want to run it within another behavior. Because of the second desire, BehA has KeyA declared as an 'Input Key' in the State Machine Interface box. Then, in my other behavior named BehB, that uses BehA, I have to again set a default value for KeyA in the State Machine Userdata box. This value overrides the value set in the state machine definition for BehA when running BehB (good). 

This is fine for when I actually want to override a userdata input key value. But lets say the BehA has 10 userdata input keys, and sometimes I want to override 1 key but other times I want to override all 10. Declaring all 10 keys of BehA as behavior input keys is fine. However, I think the **requirement** to override their default values in the State Machine Userdata of BehB is both unnecessary and annoying. 

I think a much better approach would be that if a key is declared as a behavior input key, but has no value passed to it from 'above' then it should use its default value according to its own behavior.

Apart from this one small annoyance I have to say that FlexBE is a great piece of kit and is working well for us!

pschillinger, 2017-09-30:
Thank you for the feedback! I agree that this requirement gets a bit annoying for more input keys.

The reason that is like this is primarily the fact that an embedded behavior is, during execution, nothing else than an ordinary state machine with the content from the respective behavior. Thus, the same requirements apply. But on the other hand, all information would be there to handle this requirement internally.

I will look into this and let you know.

pschillinger, 2017-09-30:
Behavior input keys can now be checked as "default" to indicate that the default value should be used instead of overwriting the value.

Let me know if this properly addresses what you had in mind. Please note that I pushed the changes only for the new [FlexBE/flexbe_app](https://github.com/FlexBE/flexbe_app/tree/develop) (and there on the develop branch). The version number should say 2.0.4 or higher.

tcassell94, 2017-10-11:
Thanks @pschillinger. I have implemented your updated app and can say I'm very happy with the result. FlexBE going well for us at the moment. Will close issue. Cheers

closed 2017-10-13 by pschillinger