team-vigir/flexbe_behavior_engine #39: Question: Dynamic outcomes

reinzor, 2017-10-13:
Is it possible in someway to specify dynamic outcomes as would be possible in a normal SMACH state?

e.g.

`outcomes=[str(e) for e in range(0,10)]`

pschillinger, 2017-10-13:
In some way, yes. From a state-perspective, outcomes can be set to the value of one of the state parameters, see for example the [DecisionState](https://github.com/team-vigir/flexbe_behavior_engine/blob/master/flexbe_states/src/flexbe_states/decision_state.py#L31).

However, from a behavior-perspective, the value of this parameter is somewhat limited. It is only possible to use values which the GUI can resolve to a list of strings. [The code which does that](https://github.com/FlexBE/flexbe_app/blob/master/src/_helper/varsolver.js) is one of the older parts of the app and could use an update, I guess. It is able to resolve variables defined under "Private Configuration", but it wouldn't evaluate Python code, although the generated behavior itself could use this mechanism.

Given by the additional capabilities of the new flexbe_app framework, it should be possible to actually evaluate Python expressions at this point. I can look into this during the next days and let you know. Would it be sufficient for what you have in mind to require that the Python expression stands by itself, i.e., does not use any variables defined elsewhere?

dcconner, 2023-05-17:
any further discussion?