team-vigir/flexbe_behavior_engine #40: Question: Textarea edit for strings

reinzor, 2017-10-13:
We would like to add a parameters using a textarea (https://www.w3schools.com/tags/tag_textarea.asp) is this possible?

If not, I would like to add this to the GUI, any pointers?

Thanks,

Rein

pschillinger, 2017-10-13:
This is not implemented, but should be quite easy to add.

You can change [this line](https://github.com/FlexBE/flexbe_app/blob/master/src/ui/ui_dashboard.js#L701) to use `textarea` instead of `input` or condition this for example on the type of the parameter: `param.type=="text"? "textarea" : "input"`. However, some adjustments might be useful such as changing the class to `"input_text"` or `max-height` to something like 50px since the flipping capability of the panel interferes with the dynamic scaling.

It's probably best if you just play around a bit until you are satisfied and make a PR if you think this change would be useful. Let me know if you have further questions on this.

I actually have one question myself: Since the parameters are meant to be used for text entered by the operator before each behavior execution, I usually experienced rather short values. What is in your application the use case for which a textarea makes more sense?

reinzor, 2017-10-15:
Thanks for the information. I will take a look at this tomorrow. We would like to specify grammars for speech recognitions, which are usually multiple lines. This also fired up the other question ( #39 ) since we would like to dynamically generate outcomes from the specified grammar. Example grammar:

```
ROOT["say_hi"] -> Hi | Hello | Hey
ROOT["say_name"] -> Could you say your name? | What is your name?
```

More info: https://github.com/tue-robotics/grammar_parser

The idea is to let the user specify a grammar in the text area box and then auto generate the outcomes specified in the ROOT semantics + the timeout of failure outcome.

dcconner, 2023-05-17:
any further discussion before I close?

closed 2023-05-18 by reinzor