team-vigir/flexbe_behavior_engine #43: FlexBE Chrome app disconnected: "InvalidServiceException: Service /rosapi/get_param does not exist"

josephcoombe, 2017-11-16:
Hi! First of all, FlexBE seems really cool. Being able to generate SMACH code from drag-and-drop states is awesome!

Unfortunately, I am unable to execute my behavior from the Chrome app. It seems to be disconnected. I'm running `ros-kinetic-desktop-full`.

[image]

----
Here are my original terminal commands:

`$ roslaunch flexbe_onboard behavior_onboard.launch`
[image]
  
`$ rosrun rosbridge_server rosbridge_websocket`
[image]

----
Revised terminal commands:

`$ roslaunch flexbe_onboard behavior_onboard.launch`

`$ rosrun rosapi rosapi_node`

`$ rosrun rosbridge_server rosbridge_websocket`
[image]

Starting the rosapi_node in a separate terminal prevents the `[ERROR] [1510868269.935593]: [Client 0] [id: call_service:/rosapi/get_param:1] call_service InvalidServiceException: Service /rosapi/get_param does not exist`
however, the FlexBE Chrome app still says it's disconnected

----
Additional info:

I ran `$ sudo apt-get install ros-kinetic-rosbridge-server` to make sure that rosbridge-server was installed & updated.

`$ rostopic list`
[image]

`$ rostopic echo /flexbe/heartbeat`
[image]

pschillinger, 2017-11-19:
Thank you for the helpful error report! The fact that the heartbeat messages are coming through, but the Chrome app stays in "disconnect" indeed suggests a rosbridge problem. I saw the rosapi/get_param-error appear before in a nondeterministic way, but usually, avoiding this error by starting rosapi separately made it work again.

Did you start the Chrome app directly (from Chrome) or did you run ``roslaunch flexbe_widget behavior_ocs.launch``? It should not cause the reported error, but in general this is required to run behaviors from the Chrome app.

In any case, could I ask you to switch to the new [flexbe_app](https://github.com/flexbe/flexbe_app) instead of using the Chrome version? There have been multiple problems with rosbridge in the past, unfortunately. The version does not use it anymore and besides this, Google ends support for Chrome apps next year, so expect the FlexBE Chrome app to become deprecated soon and the new flexbe_app will be on master.

josephcoombe, 2017-11-20:
I started the Chrome app directly from `chrome://apps/`.

Makes sense. I'll switch over the to the new flebe_app!

josephcoombe, 2017-12-04:
Got started with the new [flexbe_app](https://github.com/flexbe/flexbe_app). It connects to ROS without any warning messages!

If I run into anything odd, I'll open an issue in its [repo](https://github.com/flexbe/flexbe_app). Thank you!

closed 2017-12-04 by josephcoombe