team-vigir/flexbe_behavior_engine #44: FlexBE: Best way to generate code & pass in parameters/arguments

josephcoombe, 2017-11-16:
Thanks for making this package!

After reading through the documentation and trying the tutorial, I have a couple questions:

1) Ideally, I'd like to describe a Behavior in FlexBE's GUI and then generate a python module containing that Behavior described in SMACH (with no/minimal FlexBE dependencies).

e.g:

`$ python generatedbehavior.py`

or

```
<node name="MyCoolFSM" 
           pkg="behavior_hello_world_demo" 
           type="hello_world_demo_sm.py" 
           output="screen" 
           args="-to_camera_1_topic camera_1/publish
                     -from_camera_1_topic camera_1/subscribe" 
                     -ros_rate 100.0" 
/>
```

Is this currently supported/planned, or can you only run FlexBE generated behaviors via FlexBE?

2) If I'm running a FlexBE-generated behavior from a *.launch file, I'd want to be able to pass in parameters/arguments. Something like this:

```
<node name="behavior_launcher" 
           pkg="flexbe_widget" 
           type="be_launcher" 
           output="screen" 
           args="-b my_behavior_1" >
  <param name="to_camera_1_topic" value="camera_1/publish" />
  <param name="from_camera_1_topic" value="camera_1/subscribe" />
  <param name="ros_rate" value="100.0" />
</node>
```

or 
```
<node name="behavior_launcher" 
           pkg="flexbe_widget" 
           type="be_launcher" 
           output="screen" 
           args="-b my_behavior_1 -to_camera_1_topic camera_1/publish
                     -from_camera_1_topic camera_1/subscribe" 
                     -ros_rate 100.0" 
/>
```

This type of functionality would be useful in using FlexBE-generated behaviors in conjunction with other ROS nodes, but it doesn't seem to be supported atm. Is there an accepted work-around that I'm missing?

Keep up the good work!

pschillinger, 2017-11-19:
1. This has never been the purpose of FlexBE. But I see that it would be nice to have an option like this as kind of a final release if the runtime monitoring and operator interaction features of FlexBE are only desired for debugging. While it probably wouldn't be a big deal to add a way for executing a behavior directly without an onboard engine and without a way for the runtime features, I don't see why/how a dependency to FlexBE itself could be avoided or would make sense to be avoided. For example, the state lifecycle which includes the on_enter() etc. methods is among others provided by the flexbe_core library and usually used when implementing states, at least in my experience.

2. The following format similar to passing launch file args would work:

```
rosrun flexbe_widget be_launcher -b my_behavior_1 to_camera_1_topic:='camera_1/publish' from_camera_1_topic:='camera_1/subscribe' ros_rate:=100.0
```

or in a launch file:

```
<node name="behavior_launcher" 
           pkg="flexbe_widget" 
           type="be_launcher" 
           output="screen" 
           args="-b my_behavior_1
                     to_camera_1_topic:='camera_1/publish'
                     from_camera_1_topic:='camera_1/subscribe'
                     ros_rate:=100.0" 
/>
```

josephcoombe, 2017-11-20:
1. Cool, makes sense. If something along these lines makes its way into a final release, that'd be awesome. Otherwise, I'm not complaining :+1: 

2. Thank you! I'll try that syntax out.

josephcoombe, 2017-12-06:
2. This worked for me. Thanks.

closed 2017-12-06 by josephcoombe

tropappar, 2022-03-24:
Good to know, I added it to the documentation!

tropappar, 2022-03-24:
One important caveat: Numeric parameters are assumed to be integers by default. If you pass a floating point value, it fails.
In order to enable floats, the default value of the parameter (in the behavior XML) must be specified with decimal, e.g., 10.0 instead of just 10.