team-vigir/flexbe_behavior_engine #45: Unable to execute multiple instances of the same FlexBE behavior

josephcoombe, 2017-12-06:
I made a FlexBE behavior to control an arm.
I can run this behavior with no problems using roslaunch like so:
```
<node name="right_arm" pkg="flexbe_widget" type="be_launcher" output="screen"  
           args="-b centipede_two_dof_leg
                      side:='right'
                      shoulder_topic:='a' elbow_topic:='b' wrist_topic:='c'  
                      pos_cmd_topic:='c' " 
/>  
```
---
However, I'd like to start two instances of this behavior (one for each arm) like so:

```
<node name="right_arm" pkg="flexbe_widget" type="be_launcher" output="screen"  
           args="-b centipede_two_dof_leg
                      side:='right'
                      shoulder_topic:='a' elbow_topic:='b' wrist_topic:='c'  
                      pos_cmd_topic:='c' " 
/>  
<node name="left_arm" pkg="flexbe_widget" type="be_launcher" output="screen"  
           args="-b centipede_two_dof_leg
                      side:='left'
                      shoulder_topic:='d' elbow_topic:='e' wrist_topic:='f'  
                      pos_cmd_topic:='g' " 
/>  
```

This does not work and I get the following errors in my `roslaunch flexbe_onboard behavior_onboard.launch` terminal:

```
[ERROR] [1512581009.381507]: InvalidStateError: Attempting to close a container that is not currently open.
[ERROR] [1512581009.434119]: Behavior construction failed!
Attempting to close a container that is not currently open.
[ERROR] [1512581009.484774]: Dropped behavior start request because preparation failed.
```

---
Is this intended behavior? If yes, what would be the best way to accomplish the same effect using FlexBE

Thank you!

pschillinger, 2017-12-07:
I need to look into this and will let you know.

In general, FlexBE is designed such that the onboard engine is responsible
for coordinating the executed behaviors and always exactly one behavior is
active. Still, namespaces are supposed to work and are the recommended way
to separate independent engine instances (e.g. to control two physically or
logically separate parts of the robot).

manox, 2019-03-01:
Is this also possible with two FlexBE app instances (running on one PC) monitoring two behaviors on two robots?
The two Robots are different ROS_MASTERS and I also tried different namespaces, but if I launch the second FlexBE app on the PC, both FlexBE app instances freeze.