team-vigir/flexbe_behavior_engine #47: Question: Make use of generic behaviors from another repo

reinzor, 2018-01-03:
Is it possible to make use of behaviors that are in a separate repository? I know this is possible for states but I cannot find this for behaviors.

dcconner, 2018-01-03:
behaviors should be visible for any package that exports the <flexbe_behaviors /> tag in the package.xml.
I'm not sure when the feature was added, but it should work with the feature/flexbe_app branch of the behavior engine and latest flexbe_app instead of the original chrome app.

drazenb, 2019-11-25:
I have tried using this in order to load the behaviors from the vigir_behaviors repo, but I can't seem to make it work (I am using the flexbe_app version 2.2.1). After adding the <flexbe_behaviors /> tag, the "Load behavior" does list the vigir_flexbe_behaviors package. But it does not list the behaviors from that package. Plus, in the terminal there are messages saying that "Manifest and code need to be in the same ROS package" (one message for each behavior). Any idea how to fix that? 

(BTW, the states in vigir_behaviors are visible and can be used in the state machine editor, but only after I added in package.xml an export with the <flexbe_states /> tag.)

pschillinger, 2019-11-26:
The vigir_behhaviors repo has been created for the old FlexBE Chrome App,
i.e. FlexBE App version 1.x.

In the [readme file](
https://github.com/flexbe/flexbe_app#backwards-compatibility) of the FlexBE
App, there is a section on backwards compatibility. To use the automatic
conversion feature, make sure that no other behavior packages are sourced.

You can also do the conversion manually. This essentially means to move the
behavior source code from the individual packages to the package that
contains the manifest files and adjusting the manifests accordingly. Let me
know if you need further assistance.

>

drazenb, 2019-11-27:
I managed to do it manually. Thanks for the quick reply!

closed 2023-05-17 by dcconner