team-vigir/flexbe_behavior_engine #50: Default arguments for states?

Rayman, 2018-02-21:
Is it possible to make a state with parameters that have a default value? I could not determine this was possible based on the documentation. 

On the python side you can use default values, but how to specify this in the documentation string such that the flexbe app allows this?

pschillinger, 2018-02-22:
Yes, this is possible. Setting the default argument in Python is already
enough, you don't need to do anything else in addition. When the App parses
a state, it recognizes Python defaults.

If this is not what happens for you, please let me know some more details.

Rayman, 2018-02-27:
I'm still using the old chrome app, maybe that could be the reason. Is the `flexbe_app` almost ready?

dcconner, 2018-02-27:
You need to use the correct branch for flexbe_behavior_engine, but we've been using flexbe_app for a  while.

closed 2019-10-11 by Rayman