team-vigir/flexbe_behavior_engine #54: Did not find behavior with requested name

SDBonhof, 2018-05-07:
I just reinstalled flexbe after having to reinstall Ubuntu. I configured the widget, found my behaviors, but as soon as I try to run it i get the following error message:
[ERROR] [1525697270.562160]: Did not find behavior with requested name: Speed_challenge

I tried to make a new behavior to test whether I can run new behaviors. I can successfully save the test behavior, but i get the same error message when I try to run it.

I have already tried sourcing my setup.bash and catkin_make but to no avail.

SDBonhof, 2018-05-07:
I managed to fix it. For anyone getting this error as well:
If you already have a repo for flexbe: use the rosrun flexbe create_repo [project_name] command with your project name being the name of the existing repo **without** _behaviors at the end. 

Also, make sure that the workspace this repo is in is properly sourced in all terminals.

tiralonghipol, 2018-06-25:
hey! how and where did you set the path in order to find the created behaviors inside the app? 
Thanks

closed 2023-05-17 by dcconner