team-vigir/flexbe_behavior_engine #63: Can't connect FlexBE from different systems

utkarsh447, 2018-11-15:
Hi, I am following FlexBE tutorials. The issue is I can't connect the flexBE from different systems. My goal is to run FlexBE in one system and when behaviour is launched it should work on my Pi and show updates on my system.
I am using ros-bridge in my Pi 
using command 
`rosrun rosbridge_server rosbridge_websocket` 
and i get this output with one error when client is gets connected.
[image]

But the system on which i want the FlexBE to run, I get this,
on executing this command, 
`roslaunch flexbe_onboard behavior_onboard.launch`
[image]

Here I got connected from ROS config.
But i can't find any means to run the behaviour. 
Please suggest suitable way if i am missing some steps or making a mistake?

pschillinger, 2018-11-15:
There might be two problems here, actually:

* The first screenshot appears to be a race condition within rosbridge, I have seen it previously in several cases.
* As it looks like in the second screenshot, you did not yet add any states and behavior packages. Did you follow the instructions here for installing the Chrome app?: http://philserver.bplaced.net/fbe/download.php

In any case, I can recommend to switch to the newer FlexBE App instead of the Chrome app, see #46. This will give you several improvements including that both of the above issues are not relevant for the FlexBE App anymore. Is this an option for you?

utkarsh447, 2018-11-16:
Hi, Thank you for response, I followed your steps and got the new FlexBE app working, But still I can't reach my solution.
First I setup working Master Slave connection between two systems.
In Master System, I got connected by rosbridge
[image]

And testing the "Example Behaviour" sample program
In Master, I ran `roslaunch flexbe_app flexbe_ocs.launch` , then loaded the "Example Behaviour" code. 
and on Slave, I ran `roslaunch flexbe_onboard behavior_onboard.launch`

On Master, After running, I get the output as 
[image]

I get this message whenever I click on "Start Execution".
Also, at slave I get the same error:
[image]

Please tell me the reason for this error, and how can this be solved?

utkarsh447, 2018-11-16:
> Hi, Thank you for response, I followed your steps and got the new FlexBE app working, But still I can't reach my solution.
> First I setup working Master Slave connection between two systems.
> In Master System, I got connected by rosbridge
> [image]
> 
> And testing the "Example Behaviour" sample program
> In Master, I ran `roslaunch flexbe_app flexbe_ocs.launch` , then loaded the "Example Behaviour" code.
> and on Slave, I ran `roslaunch flexbe_onboard behavior_onboard.launch`
> 
> On Master, After running, I get the output as
> [image]
> 
> I get this message whenever I click on "Start Execution".
> Also, at slave I get the same error:
> [image]
> 
> I also got heartbeat:
> [image]
> 
> Please tell me the reason for this error, and how can this be solved?

Hi, I solved this issue, My code wasn't on the slave system. After pasting the code, it worked.

utkarsh447, 2018-11-16:
> There might be two problems here, actually:
>
> * The first screenshot appears to be a race condition within rosbridge, I have seen it previously in several cases.
> * As it looks like in the second [...]

Hi, @pschillinger , Thank you for responding in short amount of time.
In your reply, You mentioned about the new [FlexBE App](https://github.com/FlexBE/flexbe_app), I successfully ported on that and able to use it. 

In the new App, Is it possible to run the selected behavior from command line instead of manually selecting them from the GUI?

pschillinger, 2018-11-16:
Great to hear that it works now!

Yes, it is possible to run behaviors without the user interface, either
from command line/launch file or as action calls. See the following
tutorial for details:
http://wiki.ros.org/flexbe/Tutorials/Running%20Behaviors%20Without%20Operator

As mentioned there, you can also attach the user interface to an already
running behavior that you launched without it, e.g., for debugging. When
using the new FlexBE App, the launch file is:
roslaunch flexbe_app flexbe_ocs.launch

utkarsh447, 2018-11-19:
Thanks @pschillinger , it is working as it should. The only difference is I have to run the flexBE app in super user mode and then running the behavior in my Pi, I was able to see the changes in my main PC.

utkarsh447, 2018-11-19:
Hi @pschillinger , I am facing another Issue,
I have made a gif of that, please refer to the below gif:
[image]

The flexBE app is running in my remote PC using command: `roslaunch flexbe_app flexbe_ocs.launch` and in my pi, the flexBE onboard is running using command: `roslaunch flexbe_onboard behavior_onboard.launch`.

As you can see, there are 4 states in **StateMachine Editor**, First State displays "Hello World", Second state has 5 second wait timer, Third state displays "End of FSM" and the fourth state also has 3 second timer. Also, the autonomy level of all states are set to "High" and **Block Transition which require atleast "Low" autonomy** is set.

But on executing the behavior using `rosrun flexbe_widget be_launcher -b 'behavior_fsm_drone'`, You can see the states are not updating but logs are coming as it should. I am getting logs on Pi Onboard too.

Please suggest me the issue i am facing and how it can be solved?

pschillinger, 2018-11-24:
I need to check further, but I think what happens is the following.

There are two different ways to run a behavior: (1) in interactive mode using the app and (2) in fully autonomous mode, e.g., by using the `rosrun ...` command.

It looks like those two different ways are mixed in your case. When you start the behavior, it runs in mode (2) and therefore, has the shared autonomy disabled. Still, the app appears to recognize the behavior start and directly monitors the execution.

What should happen is that the app notifies you that a behavior has started and asks if you want to attach. Before attaching, mode (2) is active and upon attaching, it will switch to (1), including the shared autonomy.

utkarsh447, 2018-11-24:
Hi @pschillinger , I had it solved this morning.
I was using flexBE with ssh added on it. To get the files to be modified on different systems, i had to run it in super user mode. Whereas the onboard on the bot was running on the normal mode (which was the root of this issue). On giving the onboard the super user access, it got solved.
I want to thank you for showing your concern and providing support.

closed 2019-10-22 by utkarsh447