team-vigir/flexbe_behavior_engine #75: Behavior execution failed with AttributeError

amrp1, 2019-07-03:
I am using Ubuntu 16.04 and ROS Kinetic.I am trying to access a a nav_msgs/Path message passed through userdata in another state which accepts this '_external_path' as an input key.However i get this error  

```
[9:26:36 AM] Onboard engine just started.
[9:26:43 AM] --> Starting new behavior...
[9:26:54 AM] No behavior active.
[9:26:54 AM] Behavior execution failed! [-]
    Could not execute state 'Pre goal check' of type '': Traceback (most recent call last):
      File "/home/agv/catkin_ws/src/executive_smach/smach/src/smach/state_machine.py", line 247, in _update_once
        self._remappings[self._current_label]))
      File "/home/agv/catkin_ws/src/flexbe_behavior_engine/flexbe_core/src/flexbe_core/core/event_state.py", line 54, in _event_execute
        self.on_enter(*args, **kwargs)
      File "/home/agv/catkin_ws/src/flexnav_behaviors/flexnav_flexbe_states/src/flexnav_flexbe_states/check_goal_state.py", line 70, in on_enter
        self.new_path = userdata._external_path
      File "/home/agv/catkin_ws/src/executive_smach/smach/src/smach/user_data.py", line 165, in __getattr__
        return object.__getattr__(self, name)
    AttributeError: type object 'object' has no attribute '__getattr__' 
```

My understanding is that there seems to be some error in the smach/user_data.py file, but this is the standard version of smach that I'm using from source.Was i supposed to make any changes to smach itself or is this a bug?

pschillinger, 2019-07-03:
It looks like this error results from:

flexnav_flexbe_states/check_goal_state.py", line 70, in on_enter
        self.new_path = userdata._external_path

Did you declare "_external_path" as input key? You can only access it for
reading if it has been declared. Otherwise, it will throw an AttributeError
just as you observed here.

amrp1, 2019-07-03:
Yes i did include it as an input key, which is why this is so confusing.

pschillinger, 2019-07-03:
I see, then this is indeed confusing.

Is there any chance that you can share the source code of this state or the
on_enter method?

For completeness: I also use standard smach, there should be no need to
modify anything.

amrp1, 2019-07-03:
Sure , I do not have access to the code right now but can upload it within 12 hours

pschillinger, 2019-07-03:
I just noticed in the smach source code in the line before the one which
raises the error:

if name[0] == '_':
return object.__getattr__(self, name)

There seems to be a different treatment for keys starting with _. Thus, it
should probably be avoided for userdata keys.

amrp1, 2019-07-04:
Thanks, your last comment led me in the right direction. The leading '_' had to be removed from the input key name. It works now. :)

closed 2019-07-04 by amrp1