team-vigir/flexbe_behavior_engine #76: Question: Behavior Tests Clarification

Magnusgaertner, 2019-07-03:
I'd like to write some state_tests but for a flexbe_behavior not a flexbe_state. Is this possible and how?

pschillinger, 2019-07-03:
Unfortunately, it is not yet possible to use flexbe_testing with behaviors.
However, I don't see any conceptual problem to add this feature and would
like to do so eventually. I can add this as one of my next tasks.

Magnusgaertner, 2019-07-03:
For now I added the following to the behavior to be tested:

```
	# [MANUAL_FUNC]
	def on_start(self):
		pass
	def execute(self, userdata = {}):
		self.prepare_for_execution(userdata)
		return self._get_state_machine().execute()
	
	def on_stop(self):
		pass
	
	# [/MANUAL_FUNC]
```

which seems to work

Magnusgaertner, 2019-07-03:
another try:

	def execute(self, userdata = {}):
		return self._get_state_machine().execute(userdata)

pschillinger, 2019-09-16:
Just as a brief update, I started to refactor the testing framework with the goal to add behavior testing afterwards. Progress is tracked on this branch: [feature/test_behaviors](https://github.com/team-vigir/flexbe_behavior_engine/tree/feature/test_behaviors)

Achllle, 2020-02-05:
Seems like this can be closed @pschillinger ?

pschillinger, 2020-02-09:
Indeed, it can. I just wanted to wait for the next melodic release so that it is actually included in the installed binaries. This happened last Thursday.

closed 2020-02-09 by pschillinger