team-vigir/flexbe_behavior_engine #78: subscriber_state not getting messages even though they are published.

galbra-rnd, 2019-08-28:
Hello,
I have a mavros running, that publishing "/mavros/state" topic..
[image]

And i have a state subscriber_state (i've called it 'disarm'), that subscribes (supposedly) to this topic:
[image]
Where "safety_stopic=/mavros/state".

But, when i run this behavior, this state enters to its executing function, and return with empty hands, because this condition
```
if self._sub.has_msg(self._topic) or not self._blocking:
```
is false, even though that topic is being published for sure:
[image]

What am i missing here?

Thanks Gal.

pschillinger, 2019-09-16:
Thanks for reporting! However, I have not been able to reproduce the observed behavior. Can you run the following:

    roslaunch flexbe_states run_tests.launch

and check whether the `subscriber_state_pose.test` fails as well?

galbra-rnd, 2020-03-03:
Hey, sorry for the delay.

It is not relevant anymore.
Gal

closed 2020-03-04 by pschillinger