team-vigir/flexbe_behavior_engine #81: ProxyActionClient : Issues with sending custom goal

RajArdeshna, 2019-09-20:
Hi,

I have a custom Action Server written in C++ where i just have to send the 'Proportional' value in order to perform the action. In SMACH I implemented this state using the following example and its working fine.
I am passing my proportional value as "goal_slots=['p']" in SMACH state

[image]

But, looking at the 'ProxyActionClient' implemented in FlexBe, as per my understanding, only sending the 'Fixed goal message' is possible (Correct me if i am wrong) using the function 
                                                  client.send_goal(topic,goal) 
as described in the following example.

[image]

So, is there a way to send just the 'Proportional' value to my custom written Action server using FlexBe?

pschillinger, 2019-09-21:
It is also possible to construct an action goal from userdata input keys.
In general, this tutorial might be helpful for you:
[Developing Actionlib States](
http://wiki.ros.org/flexbe/Tutorials/Developing%20Actionlib%20States)

RajArdeshna, 2019-09-23:
Thanks for the quick response. This tutorial solved my problem :)

closed 2019-09-23 by RajArdeshna