team-vigir/flexbe_behavior_engine #83: Problems with "Running Behaviors Without Operator"

RajArdeshna, 2019-10-18:
Hello , 

I have created the behavior using the FlexBe GUI and i am able to run my behaviors successfully using GUI. Now, i want to run this behavior in a fully autonomous mode. So, i followed the steps from this tutorial : http://wiki.ros.org/flexbe/Tutorials Running%20Behaviors%20Without%20Operator
But, i am not able to run my behaviors.
My FlexBe editor looks like this:
[image]

I am able to launch the command 'roslaunch flexbe_onboard behavior_onboard.launch' successfully.
[image]

When i launch the command rosrun flexbe_widget be_launcher -b 'SAGV' , i cannot see anything happening
[image]

Moreover, i have tried launching the rosrun flexbe_widget be_launcher -b 'SAGV'  from the other computer connected with ROS master as well as the same computer which runs the onboard behavior engine on the robot.
Could you please help, what am i doing wrong here ?

pschillinger, 2019-10-18:
In terms of FlexBE usage, the commands you describe look correct. Just note that, on the screenshot, you forgot the "-b" and thus, no behavior will be started. As you correctly wrote, the command should be `rosrun flexbe_widget be_launcher -b 'SAGV'`. Did you check the ROS setup to make sure that ROS_IP or ROS_HOSTNAME are set in all terminals?

Just as a note, running a behavior without operator means that the GUI will not monitor the behavior, i.e., the FlexBE App should offer you to attach to the running execution, but it will not show the behavior execution in Runtime Control automatically.

RajArdeshna, 2019-10-25:
Such a silly mistake !
Thanks for pointing it out. Everything works now.

closed 2019-10-25 by RajArdeshna