team-vigir/flexbe_behavior_engine #85: Question: support for roscpp states/nodes

crichardson332, 2019-11-12:
Is there currently any way to incorporate existing c++ nodes (states) into a flexbe behavior? What would it take to make that work? 

My company is looking at replacing our existing implementation of smach with FlexBE, but our system is mostly c++ and our algorithms are all c++ libraries (they have to be for efficiency - we do a lot of machine learning perception stuff, and generally a lot of robotics libraries out there are only available as c++ libraries). However it would not be worth it if for every state in our system, we had to create a proxy python state in FlexBE and do a bunch of pubsub between our nodes and the python nodes.

dcconner, 2019-11-12:
FlexBE works with Python nodes, but of course Python can call C++ functions.

Just write a simple wrapper for your C functions.

crichardson332, 2019-11-12:
Is there any plan to support roscpp in the future? Wrapping our entire codebase and all our many libraries with python is not feasible, and we already have a bunch of behaviors written as C++ ros nodes anyway. We also work with contractors who often send us ros nodes to integrate in our system, and they're usually C++ nodes. 

I've seen documentation about creating action states using actionlib - is it possible for a node that uses actionlib to be integrated into FlexBE? All of our behaviors use actionlib so that could potentially work.

dcconner, 2019-11-12:
Yes.  The "states" should only be simple states that run quickly.   For more complex calculations, we usually create a "action interface" that ties to the more computationally intensive calc.  For example, you can see my "Flexible Navigation" paper https://ieeexplore.ieee.org/document/7925266

crichardson332, 2019-11-12:
Okay I'll take a look at that. Thanks for your help.

closed 2019-11-12 by crichardson332