team-vigir/flexbe_behavior_engine #86: Question: runtime configurability over radio

crichardson332, 2019-11-12:
Is there any way to achieve runtime configurability of a behavior over a wireless communication channel? For example, if the behavior engine were running on a wireless UAV and communication is only possible over a radio with spotty/imperfect comms, is there any way to configure a behavior while the vehicle is in flight? Using gui programs would not be possible in this scenario, only simple ssh terminal sessions can run over this radio link.

dcconner, 2019-11-12:
In many ways this was similar to our initial use case in the DARPA Robotics challenge with the constrained bandwidth communications between our Operator station and onboard robot. That was the primary motivation for the OPS station/Onboard division and the state machine "mirror" that we use.

I suggest you read one of the Team ViGIR DRC papers (at least one is open access) for more information.

Behaviors can be edited and changes sent via basic ROS messages.  We had a custom interface to transmit the ROS messages across our spotty link.  We had one ROS master for OPS and one for onboard, and a custom "bridge" to communicate.

crichardson332, 2019-11-12:
Ah okay, that could be a solution. We can certainly transmit serialized ros messages over the radio. Thanks!

closed 2019-11-12 by crichardson332