team-vigir/flexbe_behavior_engine #87: execute() is called just once for the first state of a concurrent container

benmaidel, 2019-11-18:
The `execute()` function of the concurrent's first state (the one where the initial state dot is connected to) is called just once. 

Example root:
[image]

Example concurrent container:
[image]

The concurrent container does have three wait states. `wait_1` which will wait for one second. `wait_2` wait's for two seconds and `wait_3` for three seconds. The expected behavior would be a `finished` outcome of the container. `wait_1` should result in `done` after one second and the concurrent container should preempt the other two wait's. But that is not the case. After two seconds `wait_2` finishes and triggers the aborted outcome of the concurrent container.

I could track the problem a bit down and believe that the bug was introduced in https://github.com/team-vigir/flexbe_behavior_engine/commit/f4bf9818b9c5a6d3f8c157d14767e521cec688a4. First concurrent state's remaining-sleep-duration will always be above 0 after the first run. So the execute will never get called again...

[test_behaviors.tar.gz](https://github.com/team-vigir/flexbe_behavior_engine/files/3859976/test_behaviors.tar.gz)

pschillinger, 2019-11-19:
Good catch, thanks for reporting!

The bug was indeed introduced by the commit you mentioned and appeared to be caused by the fact that sleep mechanism of the concurrency container was called incorrectly. In fact, in your example, always the sleep function of the rate object of the first state is called, irrespectively of which state executes. That's the reason why the first state always had a positive sleep time.

Can you take a look at the PR #88 and verify whether the fix works as you would expect? Thanks!

benmaidel, 2019-11-20:
> Can you take a look at the PR #88 and verify whether the fix works as you would expect? Thanks!

Thank you for the quick solution. I can confirm that the fix works as expected.

drazenb, 2019-11-27:
I ran into the same issue yesterday and I can also confirm the fix above works.

LoyVanBeek, 2019-12-03:
Works like a charm. Issue can be closed I think

closed 2020-03-31 by pschillinger