team-vigir/flexbe_behavior_engine #92: flexbe_logs

joeaaron, 2019-12-10:
hi, I wonder where can I find the log files.

If I run the command rosun flexbe_widget evaluate_logs, nothing happens.

pschillinger, 2019-12-13:
Logging of state transitions is disabled by default to not spam the file system. In order to enable it, set the respective argument in the launch file of the onboard engine:

    roslaunch flexbe_onboard behavior_onboard.launch log_enabled:=True

Each behavior execution will then create a new file in the log folder, by default `~/.flexbe_logs`.

Also note the recent PR #90 to fix the help text of the `evaluate_logs` script.

joeaaron, 2019-12-16:
OK, Thanks for the reply.
Then,  another question:
Container has 3 types, I can not really understand Prority, is there any examples?

pschillinger, 2019-12-18:
You can consider a Priority container as an interrupt routine inside a Concurrency container. More generally, whenever a Priority container is entered, no other states will be active outside of this container (their "pause" events will be triggered and "resume" when the Priority container is exited). Other than that, it is the same as a Statemachine container.

joeaaron, 2019-12-24:
Thanks for your kind reply.
I've watch the video **Use userdata key remapping to pass runtime data between states.**
But it's too complicated for me. If I just want to use the CalculationState to accomplish this. How to make it?