team-vigir/flexbe_behavior_engine #96: Question: exchange data between two concurrent states using input and output keys

HasanFarag, 2020-01-07:
Why is not possible to exchange data between two concurrent states using input and output keys in Flexbe? Is it not appropriate from behavior-design perceptive or upgrade to be made?

p.s:  I know I can use topics for that

pschillinger, 2020-01-07:
This current implementation resembles more closely the original behavior of smach.

While I don't have a strong opinion on whether userdata should be shared between concurrent states and can imagine its usefulness in some cases, I also appreciate the determinism of each state getting its own copy of the userdata and only writing on exit.

What is your use case here for which it would be helpful?

HasanFarag, 2020-01-08:
First of all, thanks for providing us with that powerful tool as well as the quick response! My use case is using sensor values (e.g humidity , temperature, etc) to observe the surroundings and determine a required action such as specific motion to be done by the robot while the sensor is still observing! My initial implementation is states for getting values from sensors concurrently and there is a need to take action in parallel depending on these readings!

 In addition, having user-data shared between concurrent states is very useful in other cases including the case of universal supervisor to autonomously observe the states, its progress and timeouts for taking actions!

cheffe112, 2020-02-02:
The first use case you described looks like a typical ROS way of implementing the sensor callbacks in multiple concurrent states and taking action based on these inside the states' execute() functions. I don't think passing along the sensor data between the states as userdata is the right way.

In case you need multiple sensor data per action (= per state), what you could use is the Time Synchronizers in message_filters (http://wiki.ros.org/message_filters#Policy-Based_Synchronizer_.5BROS_1.1.2B-.5D) to obtain one callback for several synchronized sensor messages.

closed 2020-03-08 by HasanFarag