team-vigir/flexbe_behavior_engine #98: __init__ list: don't add inline comments

Achllle, 2020-01-14:
I couldn't figure out why my `input_keys` and `output_keys` weren't getting loaded and it seemed like updates to the code didn't affect the states even after rebuilding the workspace, sourcing, and relaunching the app.

My state `__init__` had something like this:
```
super(TheGreatestState, self).__init__(
	outcomes=['finished', 'failure'],
	input_keys=['dummy_out_key'],  # should dummy_out_key2 get added here?
	output_keys=['dummy_out_key'])
```

Finally I realized 'hey I wonder if the comment messes things up'. And when I deleted it, the app immediately updated to include the input and output keys!

AFAIK this shouldn't cause a problem, and at least should throw a warning or error. To check my own sanity I created a dummy class+method to test this and it worked fine with comments included.

pschillinger, 2020-01-19:
Thanks for reporting! It looks like this case is not accounted for by the regex-based state parser of the app. I will work on fixing this.

Recently, I added a new alternative state parser that you can manually select in the configuration. It is based on Python instead of regex expressions, so this issue should not show there. If you want, you can switch the state parser and try again in the meantime until this is fixed.

Achllle, 2020-01-19:
Makes sense, I didn't think about that. The fix is lower priority imo than a parser error.

pschillinger, 2020-02-01:
I tried to add a comment myself and was actually not able to reproduce this issue, the state was still displayed correctly. Indeed, the example above matches the pattern:
https://regex101.com/r/TldmoK/1

Is there anything else that changed and could be related? Also, I noticed that the regex requires a linebreak in the end, so the following is not recognized:
```
super(TheGreatestState, self).__init__(
	outcomes=['finished', 'failure'],
	input_keys=['dummy_out_key'],
	output_keys=['dummy_out_key'])  # comment here should not be an issue, but it is
```
This I will fix already.