team-vigir/flexbe_behavior_engine #99: Python parser and Add State button

Achllle, 2020-01-23:
The `Add State` button in the Statemachine Editor doesn't open the menu in the latest version of the app (2.2.3) when the parser is set to Python.

## Steps to reproduce:

1. `roslaunch flexbe_app flexbe_full.launch`
2. create a new behavior
3. in configuration, set `State parser` to `Python` and hit `Force Discover`
4. back in the statemachine editor, hit `Add State`

## expected result

The `Add New State` menu pops up with a list of states to select from

## actual result

Nothing happens and no error message on the terminal

## Additional notes

The original RegEx parser does work. After switching and hitting `Force Discover`, the `Add State` button functions as expected again

pschillinger, 2020-02-01:
I cannot reproduce this in general. However, I saw such a behavior before depending on the parsed states. Could flexbe/flexbe_app#51 be related, i.e., does your workspace contain a state without documentation? In any case, errors like this need to be detected and handled properly.

cheffe112, 2020-02-02:
I've had the same issue on Friday, even after my fix in https://github.com/FlexBE/flexbe_app/pull/51. Started investigating, but didn't get to the bottom yet. The only thing I found so far is that "UI.Panels.AddState.show()" is simply not executed sometimes. Hopefully I can be of more use the next days, will try to reproduce and report.

cheffe112, 2020-02-06:
So at least I've found out how to reproduce this strange bug (though not how to handle this in the FlexBE app).
This was my state which didn't crash the Python parsing process, but rendered the AddState button unusable:

```
class TimeoutState(EventState):
    def __init__(self, outcomes, timeout=None, input_keys=[], output_keys=[], io_keys=[]):
        if 'failed' not in outcomes:
            outcomes.append('failed')
        super(TimeoutState, self).__init__(outcomes=outcomes, input_keys=input_keys, output_keys=output_keys, io_keys=io_keys)
```

The issue comes from
```
        if 'failed' not in outcomes:
            outcomes.append('failed')
```
which, after removal, didn't cause the error anymore.
So I guess it's my own fault that I provoked this error with my strange FlexBE usage. :smile: 
@pschillinger Maybe you have an idea how to handle this and throw an error?
@Achllle Maybe this error is spawned by something similar in your case?

Achllle, 2020-02-06:
Interesting that this could be due to one particular state. I figured this was a bug in the app introduced in the latest version. Let me try and see if I can reproduce.

pschillinger, 2020-02-09:
Thanks for finding a way to reporduce this @cheffe112 ! I have an idea why this does not work, although I need to check how this leads to the state panel not showing up anymore.

pschillinger, 2020-02-11:
As suspected, the Python parser indeed failed by parsing the above code example due to the way I inject code for determining the interface. I found and fixed the reason that lead to not showing the "Add State" panel anymore. Now, only this particular state fails with a warning message.

In general, I would recommend that no code should be put before the `super` call in `__init__`. Still, the following modification would make the above example "work":

    if isinstance(outcomes, list) and 'failed' not in outcomes:

However, "work" here only means that the state will be parsed and listed. There is no way yet that the App would know about the "failed" outcome being added implicitly.

In this regard, the Python parser shows still the same behavior as the Regex parser: outcomes, input keys, and output_keys can be set to one of the state parameters, but anything more advanced is not supported. It also does not seem trivial to me to change this.

@Achllle does the latest develop branch help? If yes, there should now be a warning pointing you to the state that caused the issue, it would be interesting to see the reason for failure.

cheffe112, 2020-02-12:
Works like a charm; thanks a lot, @pschillinger!
The fun thing is that, using the added warning message I was able to debug my state definition so that even the app is able to parse outcomes added automatically in the constructor.

Just FYI, if you're curious (and for further debugging or adding new features), that would be something like 
```
class TimeoutState(EventState):
    def __init__(self, *args, **kwargs):
        # add failed outcome if not present (see PreemptableState)
        _failed_outcome = 'failed'
        if len(args) > 0 and type(args[0]) is list:
            # need this ugly check for list type, because first argument in CBState is the callback
            if _failed_outcome not in args[0]:
                args[0].append(_failed_outcome)
        else:
            outcomes = list(kwargs.get('outcomes', []))
            if _failed_outcome not in outcomes:
                outcomes.append(_failed_outcome)
            kwargs['outcomes'] = outcomes
        super(TimeoutState, self).__init__(*args, **kwargs)
```

Achllle, 2020-02-12:
I cannot reproduce the original problem anymore. I apologize for not having dug into it deeper. I definitely wasn't doing anything like @cheffe112 was doing though.

I did have a state that was missing the description, but I can't reproduce the original issue by deleting the description line. I suggest we close this for now and reopen if someone runs into this again.

Perhaps related but probably a separate issue: If you delete the entire class docstring, including the `'''`, the python parser get stuck whereas the regex parser does fine.

closed 2020-02-12 by Achllle