team-vigir/flexbe_behavior_engine #101: Feature request: logdebug and log level toggling

Achllle, 2020-01-24:
Aside from loginfo, logwarn, and logerror, a logdebug would be useful, similar to how ROS' logging works.
Along with this, another feature similar to autonomy levels would be useful: The ability to set the log level (e.g. only logwarn and above) on a behavior-wide level and on a state-wide level. This would be helpful for debugging states and for only seeing critical messages during normal operation

pschillinger, 2020-03-01:
I added a debug log level.

Regarding the log level per state: The purpose of this logger is rather to be independent from states. But the new state logger rework might be suitable as a wrapper to state-dependent logging (i.e., I can add a logging handler that uses the FlexBE logger). How would you like to define the log level that should be used by each of the states?

Achllle, 2020-03-02:
I was thinking of something similar to how ROS' logger works:

* you can modify the log level by selecting the level from a drop-down menu for each state/behavior/container. A default log level is encoded in the state/behavior definition which is useful for states/behaviors that are more mature versus under development. If a behavior sets the log level to info but a state contained within is set to debug, messages produced by the state logger are filtered out. 
* full logs are still written to file somewhere so that filtered messages can still be accessed.
* nice-to-have: log messages are published to `/diagnostics` or other topic so they can be viewed with `rqt_console`.