team-vigir/flexbe_behavior_engine #107: Connecting several clients to single roscore

theevilone45, 2020-02-05:
We are trying to connect flexby to remote roscore (working on another pc) after starting the behavior it stops at:
[image]
terminal:
[image]

how can we make it work?

we've allso seen an option to set a namespace in "ROS Properties" on configuration tab but it seems like this option is not active. what is the purpose of it and how can we change it?

pschillinger, 2020-02-05:
That the behavior gets stuck in the beginning can have multiple reasons.
For example:
* Is the ROS network configured correctly and can messages on both sides be
received from each other?
* Does the terminal where you launched the onboard engine print that the
behavior starts?
* Does any used state or the manual part of the behavior block execution,
e.g., because it uses `wait_for_service`?

Regarding the namespace: This is displayed as read-only. You can run ROS
nodes in a namespace for example by using a group like `<group
ns="some_namespace">...</group>` in a launch file. Just make sure that all
FlexBE-related nodes run in the same namespace and that the topics required
by your behavior are found.

>

theevilone45, 2020-02-06:
Thanks for response. Issue was caused by wrong rosmaster configuration.

closed 2020-02-06 by theevilone45