team-vigir/flexbe_behavior_engine #112: Flexbe on Jetson nano

Al-khattab, 2020-02-19:
I am trying to use flexbe to control a robot
I'm using Jetson nano embedded system to control the robot.
Jetson nano uses Ros melodic and has a different architecture than normal Linux systems.
Is it possible to install and use flexbe on the Jetson nano?

pschillinger, 2020-03-01:
I never used a Jetson Nano myself, but I would recommend you try to install

    sudo apt install ros-melodic-flexbe-behavior-engine

on the robot and see whether this works. Installing the FlexBE App might not be possible, but I would assume you would no need it anyways on the robot (rather on a development/operator PC).

Let me know whether it works or which failure you encounter, if any.

grafoteka, 2021-05-12:
Hi @Al-khattab, did you successfully install it?

closed 2023-05-17 by dcconner