team-vigir/flexbe_behavior_engine #115: Integrate Data Input for Behaviors?

ZakariaChekakta, 2020-03-09:
Hi,
I have been working recently for designing robots behavior using Flexbe, And I am wondering now if there is a way to integrate custom GUI to interact with the behavior, I have seen the Flexbe tutorials on ROS Wiki, but the advanced GUI section and integrate data inputs for behaviors are missed, So is there any current work for those topics?

pschillinger, 2020-03-22:
I guess there are multiple ways for doing this, depending on what exactly you want.

One is to simply use publisher and subscriber states and you custom GUI communication via those topics. If the interaction is simple enough, this might be very easy to integrate.

Another option is the `flexbe_input` package, which was designed exactly for the use case of providing data from a custom GUI. Originally, we used it to display user dialogs and request input from the operator such as poses, provided by an interactive marker in an RViz-like GUI. In short, this package provides an action server implementation that forwards requests between the behavior (`flexbe_states/InputState`) and an action interface of the custom GUI. Let me know in case you are interested in using this.

The advanced GUI tutorial would be on a different topic. At some point, I plan to write up various ways of advanced usage of the FlexBE App GUI. But it will not cover custom GUIs.

ZakariaChekakta, 2020-03-23:
@pschillinger Thank you for your reply, it is appreciated!

I know about the first way using simple subscriber state,
I am interested in the second way, using the flexbe_input action server with a custom GUI.
 To make it clear, let say I want to design a mobile robot behavior to navigate to several points,  Those goals are displayed on the custom GUI and the operator should choose which goal to send the robot to.

From your explanation, if I am  right, the way to implement this is by implementing a client-server so when the operator selects a goal the client will be executed

pschillinger, 2020-03-27:
Yes, your use case appears to fit best for the second way.

It's not an action client that you implement, but the server. The client-role is taken by the input state, i.e.:
* You implement an action server within your custom GUI of type `flexbe_msgs/BehaviorInputAction`.
* The goal callback of this action server is supposed to start user interaction, get the required data, pickle it, and send the pickled data back as its action result.
* At the point in the behavior where you need the data, you put a `flexbe_states/InputState`.

Then, the following happens during runtime:
* When the InputState becomes active, it sends an action goal including the data request
* Your action server and custom GUI provide the data and send it back pickled
* The InputState unpickles the data and returns it as userdata output so that it can be used by subsequent states