team-vigir/flexbe_behavior_engine #116: Crash when using Python parser and `from moveit_commander import MoveGroupCommander, PlanningSceneInterface`

LoyVanBeek, 2020-03-09:
Something crashes and hangs FlexBE app when parsing a class that has one of these classes imported directly. 

It is mitigated by importing those classes not on the toplevel of the file but where I need those classes, but still weird.

This is maybe due to those classes linking to some C stuff perhaps?

fmessmer, 2020-03-16:
mitigated....but not solved?

LoyVanBeek, 2020-03-17:
With this PR, there is a bit of a workaround. I haven't dug into exactly why it happens yet, but I think it is related to 'pyassimp' which borks on my machine.

LoyVanBeek, 2020-03-18:
Actually, this is a problem with the FlexBE app, not the behavior engine. 

Found out it due to this issue happening on my system: https://github.com/ros-planning/moveit/issues/86

Somehow my `pyassimp` install ~is~ was borked and thus when some MoveIt stuff is imported, it prints this message:
`Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info`. I've tried to fix this before to no avail. 

This also happened in the state parser, which is designed to output JSON for the FlexBE_app to ingest, but before the opening `{` of the JSON, there is the pyassimp message and thus JSON parsing fails. 

I tried setting up pyAssimp again with a newer version and the message seems to be gone.

closed 2020-03-18 by LoyVanBeek