team-vigir/flexbe_behavior_engine #117: Race condition in VigirBeOnboard

StefanFabian, 2020-03-25:
We've encountered a possible race condition in `VigirBeOnboard`.
When using the flexbe action server and spamming goals, the onboard class throws
at `active_state.name` because `active_state` is None.
This leads to the class being in a broken state where it is stuck in switch limbo and will no longer do anything.
https://github.com/team-vigir/flexbe_behavior_engine/blob/19c29cc61995497ac8bbeefd28992cca98eae769/flexbe_onboard/src/flexbe_onboard/flexbe_onboard.py#L117-L120

I can fix the error when using the action server by modifying the action server to actually wait until the behavior is finished or preempted before preempting and accepting a new goal and will submit a PR.
Nonetheless, I think that method could use some attention as it is spawned in threads and currently isn't threadsafe.

pschillinger, 2020-03-27:
Should be fixed by the above commit.

StefanFabian, 2020-03-27:
That was fast and quite the rework :)
Looks like it will fix our issue and I will check that later.
However, I still see a possible race condition.

https://github.com/team-vigir/flexbe_behavior_engine/blob/4e78ed2dfaca29527b07ccd4ed905314221ad488/flexbe_onboard/src/flexbe_onboard/flexbe_onboard.py#L89-L91
https://github.com/team-vigir/flexbe_behavior_engine/blob/4e78ed2dfaca29527b07ccd4ed905314221ad488/flexbe_onboard/src/flexbe_onboard/flexbe_onboard.py#L120-L123

It could happen that two behavior executions are started shortly after each other. and while the second one will be blocked at the switch lock, the first will execute it and if the execution of the running behavior finished, it might go right into the second lock.
At that point, the other thread may be chosen to continue and will enter the switch lock and evaluate the running variable which has not yet been updated by the first thread.
That wouldn't lead to a crash but it would mean that the first execution will not be preempted and run until it is finished or a third thread preempts it.

Not very likely to happen outside of stress testing, though.

pschillinger, 2020-03-27:
> That wouldn't lead to a crash but it would mean that the first execution will not be preempted and run until it is finished or a third thread preempts it.

True. However, just as a clarification because I guess it's not obvious: "Switch" here refers to switching to an updated version of the same behavior. The function is intended to support the "lock > change > save > unlock" workflow of the FlexBE App. It is not intended to be used as a measure to preempt behaviors.

In particular, a behavior execute command is only ever relevant for the switch function in the case that some version of the same behavior is already running. In the case you describe, I would actually argue that this is the intended behavior because it is not a switch in the above sense. If the first behavior should be preempted in favor of the second one, a "preempt" command should be sent. In fact, just commanding any other behavior than the currently running one would just be rejected if the previous behavior was not explicitly preempted before.

StefanFabian, 2020-03-27:
I've tested it and while it still doesn't behave correctly (after spamming goals most executions are preempted but when stopping the spamming the last executions get state updates but don't receive a finished result), the onboard doesn't crash anymore and requests some time after spamming goals will be handled correctly.
However, that is on the action server and a different issue. I will incorporate your feedback regarding the action server tomorrow.

>  In the case you describe, I would actually argue that this is the intended behavior because it is not a switch in the above sense. If the first behavior should be preempted in favor of the second one, a "preempt" command should be sent.

I would be sometimes preempted and sometimes not, though.
But considering that this should not happen by design anyway, I guess it doesn't really matter once the action server adheres to that design.

closed 2020-03-27 by StefanFabian