team-vigir/flexbe_behavior_engine #122: feature request: restart engine

machinekoder, 2020-04-09:
It would be very useful to be able to restart the onboard engine for testing new code. Reloading Python modules at runtime are tricky and don't work most of the time. The patch here implements this feature: https://github.com/machinekoder/flexbe_behavior_engine/tree/feature/restart
and here a button for the GUI: https://github.com/machinekoder/flexbe_app/tree/feature/restart

pschillinger, 2020-04-17:
Thanks for the suggestion! However, restarting the process as in the suggested commit appears a bit harsh to me. How about [this new commit](https://github.com/team-vigir/flexbe_behavior_engine/commit/dd5fc2d6c01459bd1e06e05eb861fd209e839a96), would this address the issue of reloading modules for you? If not, what would be missing?

smartsgic, 2021-01-29:
Can I modify the states and test in the Flexbe App immediately without relaunch everything?