team-vigir/flexbe_behavior_engine #124: Need to wait until the subscribers are ready

YehudaFrumkinGit, 2020-06-03:
We use Flexbe for building our robot. We saw that every 1 time of 5 our application comes up 
apparently correctly but the main behavior doesn't listen on the topic so the application is 
actually stuck. After debugging I think that there is a race condition between 
BehaviorLauncher (flexbe_widget/behavior_launcher.py) and 
VigirBeOnboard (flexbe_onboard/flexbe_onboard.py).

Sometimes BehaviorLauncher publishes be_selection before VigirBeOnboard subscribed to it.

This is the code of BehaviorLauncher:
		if not os.path.isfile(be_filepath_old):
			be_selection.behavior_checksum = zlib.adler32(be_content_new)
			if msg.autonomy_level != 255:
				be_structure.behavior_id = be_selection.behavior_checksum
				self._mirror_pub.publish(be_structure)
			self._pub.publish(be_selection)        
        
        
And this is the code of VigirBeOnboard:
        self._sub.subscribe('flexbe/start_behavior', BehaviorSelection, self._behavior_callback)

I am attaching the two sources with my debug tracing and I am attaching file 
flexbe_bug_reproduction.txt which is the actual trace. You can find the important points by looking for 
the label YF04 in flexbe_bug_reproduction.txt.
The first part of this file contains output of a good run when the events occurred in the right order.
The second part (after // Bad  ******) contains output of a bad run when the events occurred in wrong order.
The fix that we are going to try is to add a sleep in BehaviorLauncher

		rospy.loginfo("%d behaviors available, ready for start request." % self._behavior_lib.count_behaviors())
		rospy.loginfo("Sleep to wait for all subscribers to be ready")
		rospy.sleep(0.5) # wait for subscribers to be ready
Best regards,
Yehuda Frumkin
[flexbe_onboard.py.txt](https://github.com/team-vigir/flexbe_behavior_engine/files/4724214/flexbe_onboard.py.txt)

[behavior_launcher.py.txt](https://github.com/team-vigir/flexbe_behavior_engine/files/4724236/behavior_launcher.py.txt)

[flexbe_bug_reproduction.txt](https://github.com/team-vigir/flexbe_behavior_engine/files/4724245/flexbe_bug_reproduction.txt)

YehudaFrumkinGit, 2020-06-04:
I will be glad to hear from you. Please write to my email address yehuda@xtendrobotics.com
Best regards,
Yehuda Frumkin

pschillinger, 2020-06-05:
Hi @YehudaFrumkinGit This should actually be fixed already on the `develop` branch, see PR #120. I plan to merge this into master around the end of next week and it will be contained in version `1.2.5`.

YehudaFrumkinGit, 2020-06-07:
Thank youPhilipp,
Best regards,
Yehuda

closed 2020-06-07 by YehudaFrumkinGit