team-vigir/flexbe_behavior_engine #125: Noetic release

PaulRuvu, 2020-06-05:
Can the noetic release be expected anytime soon?

Thanks in advance.

pschillinger, 2020-06-05:
Absolutely! Python 3 compatibility should already be there since PR #102. As soon as I have my noetic installation set up and ready, I will do some final testing and adjustments where required and then do the release. I plan to do this during the next two weeks.

shendry29, 2020-10-07:
@pschillinger have you made any progress since your last post? I'm using the develop branch (to incorporate the PR you mentioned) on Noetic but am getting errors upon loading in a behavior that was loading in Melodic.

I can't paste the full traceback since the terminal within the FlexBE app doesn't allow highlight/copying, but here's the end:
```
...proxy_transform_listener.py", line 2, in import tf File "/opt/ros/noetic/lib/python3/dist-packages/tf/__init__.py", line 30, in from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException File "/opt/ros/noetic/lib/python3/dis-packages/tf2_ros/__init__.oy", line 38 in from tf2_py import * 
File "/opt/ros/noetic/lib/python3/dist-packages/tf2_py/__init__.py", line 38, in from ._tf2 import * ImportError: dynamic module does not define init function (init_tf2)
[SUB:/flexbe/heartbeat] -c:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated

Parsing sourcecode...
Code parsing completed.
Building behavior state machine...
Unable to find state definition for: initModeState
Please check your workspace settings.
Unable to find state definition for: WallFollowActionState
Please check your workspace settings.
Code parsing failed: TypeError: Cannot read property 'getOutcomes' of undefined
```

pschillinger, 2020-11-04:
The develop branch works for me on noetic although I cannot test it as much as I want at the moment (that's why not released yet). More specifically, I tried to reproduce the issue you describe, but couldn't get such an error.

To narrow it down a bit more:
* Can you confirm that you also use the develop branch for flexbe_app?
* Are the two state definitions you try to import, initModeState and WallFollowActionState, listed when you click on the Add State button?
* Are these states compatible to noetic and can you, e.g., import and instantiate them manually in a python shell?
* Which state parser are you using, Python or Regex? (see Configuration tab) What happens if you chose the other and click Force Discover?

dcconner, 2023-05-17:
Note that there is a Noetic version on https://github.com/FlexBE/flexbe_behavior_engine. 

Did you resolve the specific questions regarding behavior loading?

closed 2023-06-09 by dcconner