team-vigir/flexbe_behavior_engine #127: question about behavior_checksum and behavior_id

zengxiaolei, 2020-07-06:
Hi, I noticed the behavior_id of topic /flexbe/status  is equal to the behavior_checksum of topic /flexbe/start_behavior, but different with behavior_id of /flexbe/start_behavior. I wonder why, is this intentional?

pschillinger, 2020-07-24:
This is indeed intentional, although I agree that the naming is not perfect when looking at this particular combination. Some explanations:
- In general, there are two checksums used: (1) A checksum that is unique and consistent for each behavior and (2) a checksum unique for the actual content/source code of a behavior.
- When commanding the execution, (1) needs to be used to tell the onboard machine which behavior it should load. This would not be possible via (2).
- After loading, onboard verifies its own version of (2) depending on the actual locally loaded content with the (2) provided by the start command to ensure that the exact same version is running as expected, that's why the checksum is included in the start command.
- When running, the status topic always refers uniquely to the actually running version and thus uses (2). In particular, when modifying a behavior on-the-fly and one version of a behavior should be replaced by an update, both version have the same (1), but different (2). Thus, it is important to distinguish them by using the content-specific version as identifier.

It's a rather involved topic, but I hope this helps to clarify.

zengxiaolei, 2020-07-30:
Thanks for your reply,  it's helpful for me to understand more about flexbe!

dcconner, 2023-06-29:
The latest ros2-devel version on the github.com/flexbe/flexbe_behavior_engine site for latest development changed this to be "behavior_key" for selection, and behavior_id for the checksum consistently.

This note still applies to ROS 1 Noetic version

closed 2023-06-29 by dcconner