team-vigir/flexbe_behavior_engine #128: No vertical scrollbar

YehudaFrumkinGit, 2020-07-06:
Hello,
I put a few behaviors in a container. In our design we have a lot of behaviors that run in parallel.
But there is no vertical scroll bar in this window so we cannot add more behaviors to this container.
We can find a work around - for example, to arrange behaviors in 2 columns but it doesn't look nice.
Another possibility - split the behaviors to 2 containers that will run in parallel. I assume it would work but again it's not nice if User Interface limitation (missing vertical scroll bar) affects the logic of the program.
Best regards,
Yehuda Frumkin
I am attaching a print screen 
[image]

pschillinger, 2020-07-24:
Technically, the editor area is a drawing area, so unfortunately, I didn't find a practicable way yet to add regular scrollbars. Still, there is a solution to your issue:

When holding down the Shift key, you can scroll around by drag&drop onto the background (and Shift + Space will always teleport you back to the origin).

I know that this is not really mentioned somewhere explicitly, sorry for that! Would you have a suggestion how to add this so that you would have found it?

YehudaFrumkinGit, 2020-07-26:
Hello Philipp,
I would add a Help menu option to the menu of the Behavior editor.
Thank you very much for your help!
It solves my problem:)
Yehuda