team-vigir/flexbe_behavior_engine #129: Passing userdata to the sub behavior does not work properly on the develop branch

fgrcar, 2020-07-27:
Hi,

I tried the develop branch of this repository because of [this](https://github.com/FlexBE/flexbe_app/issues/66) and I found a bug regarding passing the userdata to the sub behavior.

So, I created a following behavior:
[image]
[image]
This behavior just takes an input userdata key and prints it out.

Then I created another behavior which uses the first one.
[image]
[image]
This behavior just prints out the key which is then passed into the sub behavior.

If I use the develop branch of this repository, the printed logs of the main behavior are:
[image]
That means that the key was not passed properly, because the default was used.
If I run the same behavior on the master, I get the desired logs:
[image]

I am not sure if I am missing something but it looks like that the passing of the userdata does not work properly on the develop branch.

Best regards, F

cjue, 2020-09-18:
There are multiple commits related to userdata on develop:
* https://github.com/team-vigir/flexbe_behavior_engine/commit/9c47ebfd18f4290340d39d6ab3b7be8c52389027
* https://github.com/team-vigir/flexbe_behavior_engine/commit/cba554eb413d6dab71ca1819a2e67f45f1c283af
* https://github.com/team-vigir/flexbe_behavior_engine/commit/9acc3c859822290b29249abda32a2a7a59b1bbb4

I also noticed another bug on develop, where userdata is not passed along between states when the behavior does not define *any* default userdata values (in the State Machine Userdata dashboard section).

pschillinger, 2020-11-04:
Sorry for my delay in addressing this, but both issues should be fixed now by the above commit. Please let me know if you still observe any unexpected behavior.

closed 2020-11-24 by fgrcar