team-vigir/flexbe_behavior_engine #131: High latency in 'behavior chains'

jmachowinski, 2020-09-04:
Hi,
we use flexbe, to control our system, and ran into an issue with chains of behaviors:
An example :
Lets say we got a chain like this :
- Check sensor input 
  - Reads some topic and passes a description on succeeded to the next state
- Get target for object
  - depending on the object return some target
- Fetch trajectory for object
  - Return a precomputed trajectory for this object type
- Execute trajectory
 - Calls an action server robot moves etc.

State 2 and 3 return instantly the first time execute is called.
What we found out now, is that the next behavior in the chain
is always called in the next execution cycle. In case of the default
cycle time this would mean, that 400ms pass, before ExecuteTrajectory
is called. 
This is a problem in our case, as we are processing objects that move
into our robot with 1m/sec.

So my question would be, is it possible to execute such chains in one execution cyle
instead of postponing state transitions into the next cyle ?
If so, were in the code is the place to implement this ?

Note, we know that we can increase the cycle rate, but this increases
CPU load significantly as we have a lot of concurrent states around.

A second workaround would be to create monolithic behaviors, but that
somehow defies the purpose of using a state machine engine in the fist place...

cjue, 2020-09-18:
You could increase the cycle rate only for the short-running states, i.e. not for the "wait for the robot to move somewhere" ones.

One way to do this would be to introduce a subclass of EventState that sets a much faster cycle rate and have specific states use that as their parent class. Or the other way around, reduce the general cycle time and have some states use a slower cycle rate.

jmachowinski, 2020-10-17:
Thanks for your suggestions.
We now switched to a different state machine implementation, so I'm closing the issue.

closed 2020-10-17 by jmachowinski