team-vigir/flexbe_behavior_engine #134: State logger configuration and question about lists as a behavior parameters data types

fgrcar, 2020-11-24:
Hi,

First of all thank you for the new release in the last week. If I'm right Smach has been removed, so I'm happy that FlexBe is evolving and improving. Keep up the great work!

Getting to the point, my first issue is regarding [this tutorial](http://wiki.ros.org/flexbe/Tutorials/Logging%20Behavior%20Execution) about the State Logger, specifically about the configuration parameters. There says that setting the handlers parameter like this `rosparam set /flexbe/log_config/loggers/flexbe/handlers [publish]` should redirect the logs from the file to the **/flexbe/state_logger** topic. As this didn't work for me, I started to dig in the code and found out that the namespace is wrong, the right one is **/behavior/log_config** and not **/flexbe/log_config**. It would be great if you could fix this in the tutorial. The next thing on the same topic which I noticed is that even though if the logs are directed to the topic, there is still a log file created which I think is a bit unwanted as it anyways remains empty.

My second issue is basically a question or a wish: would it be possible to add lists under the supported behavior parameters data types? From my experience it is quite common having robot's joints written in a list, but list could used for other purposes as well and if a behavior takes in such parameter it is currently not possible to do it directly. I've been currently solving this by simply defining the parameter as a text and then passing it to a state by using python's **eval** function. Anyway it would be much easier if I could do it directly, so I am interested in your opinion about this.

Best regards, F

pschillinger, 2020-12-06:
Thanks for pointing out the wrong namespace, you are correct. It should be **/behavior/log_config** because `flexbe_onboard.launch` names the node "behavior". I fixed this.

I could also reproduce the issue of an empty file being created when publishing logs. While this is not desirable, I'm actually not aware that I create this file. I think this could be an artifact of the Python logging system when a file handler is listed under handlers although then not being used in one of the loggers. When using the state logger properly, I would always recommend to load the log_config parameter from a system-specific yaml config file and then only specify the handlers that are actually required.

Regarding the list, I'm not fully sure how to handle it best. Maybe expect a yaml-format input string and parse it when passing to the behavior. This ensures that the types of list entries match best the expectations. On the other hand, the already available "file" type parses yaml, so the only difference would be to load from a file vs. load from user input string. Given that the original purpose of a behavior parameter is something that an operator might want to change for each behavior execution, I'm not sure how much sense it makes to expect from an operator to enter a complete yaml string previous to each behavior execution. Let me know what you think about this. If it still fills a gap, I'm happy to work on adding something in this direction.

In the mean time, as alternative to eval when passing it to a state, note that you can add a variable to "Private Configuration" where you do the list parsing once and in a central location, instead of somewhere for one or more states.

fgrcar, 2020-12-07:
Thank you for your reply.

So regarding handlers, how can I fix this unwanted file creations? I don't fully understand the answer because I think if I set the handler param just to publish there shouldn't be any file handler created or what am I missing here?

Regarding lists: yes, I've been using eval as you suggested by adding a new variable under the Private Configuration which takes in the string parameter which is a list actually and with that trick it is much easier to use it than putting eval in more states. I also agree with your statement that an operator won't be putting in large structures manually for each execution, usually I had such a behaviors as a nested ones, meaning that I was actually passing lists from a Private Configuration on the top layer to the nested behaviors as a string and then converting them there back again to lists. For me I think yaml string would not be any better than this current hack but just a plain list structure would help. Anyway, thank you for your thoughts about that topic but I think it is not that really urgent that any solution to this is added but maybe you can add this to the backlog with some lower priority if you want.

pschillinger, 2020-12-09:
The empty file issue is indeed an artifact of the native python logging module. See for example [this question](https://stackoverflow.com/questions/46251114/python-logging-to-file-through-ini-creates-empty-log-file) where the same behavior is described (although unintentional).

What I meant with my above comment was that if you don't define the file handler, also no file will be created. For example, you could create a file say `loggers.yaml`:
```yaml
version: 1
disable_existing_loggers: False
formatters: 
  yaml:
    (): flexbe_core.state_logger.YamlFormatter
handlers:
  publish:
    class: flexbe_core.state_logger.PublishBehaviorLogMessage
    topic: flexbe/state_logger
    formatter: yaml
loggers:
  flexbe: 
    level: INFO
    handlers: [publish]
```
and use it
```
rosparam load loggers.yaml /behavior/log_config
```

Then it will not create an empty file. So if the state logger is used more than just playing around to understand the principle as for the tutorial, it makes sense to create an own logging config file that only contains those handlers that are actually used. As said before, I agree that creating an empty file should not happen in the case that the file handler is not used, but changing this behavior is out of my control.

Regarding the lists, yes, I will definitely keep this in mind and I think having something in this direction makes sense. I get your point regarding nested behaviors, in this context it would indeed be useful.

fgrcar, 2020-12-09:
Thanks for your detailed explanation regarding the loggers. About the lists: great, I'm looking forward to find this some day among the new features.

closed 2020-12-09 by fgrcar