team-vigir/flexbe_behavior_engine #135: State code not updated after saving in IDE since 2.3.0

tremblay0934, 2020-11-25:
I am testing with your new release and noticed that state code modification were not taken into account after saving in my IDE. I tried changing the State parser (both python and regex), troggling cache pacakges, and force discover. Only a ROS reset takes the state changes.

I am using using ROS melodic built from source for python3 if that helps. Before updating, I had fitted your engine + GUI for python3 in my codebase (mostly addler32 mods, python3 shebangs and correct python3 executable nae in flexbe app).

Thanks for your time and efforts, me and my team are really grateful.

pschillinger, 2020-12-06:
I think this is related to [this commit](https://github.com/team-vigir/flexbe_behavior_engine/commit/bbe82cf28392ad82d09cce07cc6210579c91b6e0) where I had to revert a feature that was introduced recently before. You can comment-in those lines again and see if this solves the problem for you.

The reason why I had to remove this is that it was not working as intended for some people and there were issues with some imports when executing a behavior for the second time. I can make this feature configurable if this is useful for you, but I couldn't completely fix it yet. So there might still be problems when using it.

Just to clarify, this is not related to the GUI updating file changes, it is related to the flexbe_onboard engine when importing states (and the Python-internal caching of previously imported modules).

tremblay0934, 2020-12-06:
I uncommented the code and indeed, the desired behavior for for state changes occured. It would be good to make this feature configurable. The reason is that we use the FlexBE app for two main purpose. 
The first one is developping states, which makes the feature very handy because state code changes while we do so. 
The second reason is to execute real life robot testing. I did not encounter the bugs you mentionned yet, but if you know that disabling the clearing module feature makes the app more robust for testing, then I agree with you on adding a configuration option for this feature, at least until you have time to find fix the bugs you mentionned. 
Thanks a lot!

closed 2020-12-06 by tremblay0934

pschillinger, 2020-12-09:
You can now use `roslaunch flexbe_onboard behavior_onboard.launch enable_clear_imports:=True` (see above commit) for optionally enabling the reloading of states.

tremblay0934, 2020-12-10:
> 
> 
> You can now use `roslaunch flexbe_onboard behavior_onboard.launch enable_clear_imports:=True` (see above commit) for optionally enabling the reloading of states.

Thank you for the quick reply and feature Philipp !