team-vigir/flexbe_behavior_engine #136: /flexbe/outcome_request publishes bad indexes for the outcome list

tremblay0934, 2020-12-06:
Lets say I have a state with outcomes = ['finished', 'failed'], where finished will be triggered when an action goal succeded.

I've tested (with no autonomy on GUI) and if goal succeeds, a log messages prints : <-- Want result: test_state > finished. 
However, the topic /flexbe/outcome_request prints that the outcome value is 1, which means the index 1 of the outcomes list, which is the "failed" outcome and not the finished as it should be. The GUI tells me that Onboard requested outcome: failed and the arror for "faield" is boldened. If i click it, it goes to the other state pointed to by the "finished" outomes.

This bug happens randomly on a state and i found out that if it does not happen for your state while your are trying to reproduce the bug, you simply need to change the outcomes list in the constructor from this :
outcomes = ['finished', 'failed']
to this:
outcomes = ['failed', 'finished']

This way, i was able to reproduce the bug on any state. Same kind of behavior happens with 3 outcomes.

tremblay0934, 2020-12-06:
For the record, i use python3

pschillinger, 2020-12-09:
I assume this is indeed a Python3-specific issue. Can you check whether the above commit fixes it?

tremblay0934, 2020-12-10:
Your commit fixed the 4-5 test cases i had layed out for my application. I will mark this issue as resolved . Thanks a lot once again!

Since our robotic project will be very active for the six months to come and that it is python3 based, I shall be very active on issues/discussion related to this setup. Did you have any potential problem in mind about flexbe engine with python 3? I can definetly help you improve with documented tests.

closed 2020-12-10 by tremblay0934

tremblay0934, 2020-12-10:
I would also like to know what are your plans for merging this commit in the master, as my team of 12 devs uses an automated environment using your master branch. For example, if you don't plan to merge within 2 weeks, i will change the environment setup to switch to _develop_. 

Thank you for your feedback!

pschillinger, 2020-12-11:
I just did the merge. Since this is a quite import bugfix for Python3, I didn't want to wait for too long. Also, the release scripts are currently running so the fix should be available in the next ROS release with version 1.3.1.

> Since our robotic project will be very active for the six months to come and that it is python3 based, I shall be very active on issues/discussion related to this setup. Did you have any potential [...]

Sounds good, looking forward to new issues if you find any problems. Thanks for your offer to help! I'm not aware of any open problems with Python3, but since Python3 support is rather new, there might be some. At least the CI pipeline now also tests against ROS Noetic in Python3.